AllskyTeam/allsky #1: Having install troubles - package assumes Wheezy?

jke67, 2016-12-14:
I think my problems, some of which may be related to using the Jessie-lite image.   After installing various missing packages, I got to 

`pi@astropi:~/allsky $ ./allsky.sh 
Starting allsky camera
ls: cannot access Ortona-full.jpg: No such file or directory
entr: No regular files to watch
./capture: error while loading shared libraries: libavcodec.so.53: cannot open shared object file: No such file or directory`

 libavcodec.so.53 is current as of wheezy (while current raspian is jesse).    Will try installing with wheezy.   Although maybe a full jessie is correct way to go?

thomasjacquin, 2016-12-14:
I updated the install.sh file to pull opencv from libopencv-dev instead of the compiled version located in the dependencies folder. That should fix the error with libavcodec.so.53.

You can disregard the other errors, they will go away as soon as you capture your first image.

thomasjacquin, 2016-12-21:
@jke67 I have upgraded the code and tested it on Raspbian Lite. There is now a -nodisplay argument for distribution with no desktop environment.

closed 2017-07-04 by thomasjacquin