AllskyTeam/allsky #3: Sigill crash in RPI 2 with allsky

anismo, 2017-11-28:
I am trying to get this running on my Raspberry PI 2 board and I am trying to get it connect to my ASI120MC USB 2.0 camera.

However, following the instruction here, capture binary crashed with sigill and I realized that the Makefile is
 using -march=armv7 and my board is armv61. Once I removed it and recompiled capture, it started up fine however, the libAsiCamera2.so is probably also compiled with armv7 and is crashing with SIGILL.

Any chance I can get the libASICamera2.so build for armv6? Thanks!

==========
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xb01ac200 (LWP 1604)]

 ******************************************
 *** Allsky Camera Software v0.2 | 2017 ***
 ******************************************

Capture images of the sky with a Raspberry Pi and an ASI Camera

Add -h or -help for available options 

Author: Thomas Jacquin - <jacquin.thomas@gmail.com>

Contributors:
-Knut Olav Klo
-Daniel Johnsen
-Yang and Sam from ZWO
-Robert Wagner

Thread 1 "capture" received signal SIGILL, Illegal instruction.
0xb62c73ac in ASIGetNumOfConnectedCameras ()
   from /usr/local/lib/libASICamera2.so
(gdb) Quit

thomasjacquin, 2017-11-28:
Hi,
Can you give me the content of this command? 

`cat /proc/cpuinfo`

That will help me find what versions of the Pi are compatible.

thomasjacquin, 2017-11-28:
I have updated the Makefile and included armv6 libraries. Can you let me know if it compiles properly?
Thanks

anismo, 2017-11-28:
Hi Thomas, Thank you for your response

Here is the CPU info. I will update the makefile (I did change it and compile it already and the capture program itself works ok now - Earlier, the capture program itself hit sigill. However, now the probelm is that the shared library for the camera has the same problem . Do you also compile that library?)

pi@allsky:~/allsky $ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 000e
Serial		: 0000000094ff0400

anismo, 2017-11-28:
I just noticed that you have added new libraries. I Will build and update later today after work! Thank you

anismo, 2017-11-28:
I tried the new Makefile but it fails with the cross compiler not being available

pi@allsky:~/allsky $ head Makefile
platform = armv6

#INCLIB = /usr/local/include
#LDLIB = /usr/local/lib
OPENCV = $(shell pkg-config --cflags opencv) $(shell pkg-config --libs opencv)
USB =  -I libusb/ -L libusb/  
LIBSPATH = -L../lib/$(platform) -I../include
DEFS = -D_LIN -D_DEBUG 

CFLAGS = -g  -I $(INCLIB) -L $(LDLIB) $(DEFS) $(COMMON) $(LIBSPATH)  -lpthread  $(USB) -DGLIBC_20
pi@allsky:~/allsky $ make
arm-bcm2708hardfp-linux-gnueabi-g++  capture.cpp -o capture -g  -I  -L  -D_LIN -D_DEBUG   -L../lib/armv6 -I../include  -lpthread  -I libusb/ -L libusb/   -DGLIBC_20 -march=armv6 -lrt -I/usr/include/opencv /usr/lib/arm-linux-gnueabihf/libopencv_calib3d.so -lopencv_calib3d /usr/lib/arm-linux-gnueabihf/libopencv_contrib.so -lopencv_contrib /usr/lib/arm-linux-gnueabihf/libopencv_core.so -lopencv_core /usr/lib/arm-linux-gnueabihf/libopencv_features2d.so -lopencv_features2d /usr/lib/arm-linux-gnueabihf/libopencv_flann.so -lopencv_flann /usr/lib/arm-linux-gnueabihf/libopencv_gpu.so -lopencv_gpu /usr/lib/arm-linux-gnueabihf/libopencv_highgui.so -lopencv_highgui /usr/lib/arm-linux-gnueabihf/libopencv_imgproc.so -lopencv_imgproc /usr/lib/arm-linux-gnueabihf/libopencv_legacy.so -lopencv_legacy /usr/lib/arm-linux-gnueabihf/libopencv_ml.so -lopencv_ml /usr/lib/arm-linux-gnueabihf/libopencv_objdetect.so -lopencv_objdetect /usr/lib/arm-linux-gnueabihf/libopencv_ocl.so -lopencv_ocl /usr/lib/arm-linux-gnueabihf/libopencv_photo.so -lopencv_photo /usr/lib/arm-linux-gnueabihf/libopencv_stitching.so -lopencv_stitching /usr/lib/arm-linux-gnueabihf/libopencv_superres.so -lopencv_superres /usr/lib/arm-linux-gnueabihf/libopencv_ts.so -lopencv_ts /usr/lib/arm-linux-gnueabihf/libopencv_video.so -lopencv_video /usr/lib/arm-linux-gnueabihf/libopencv_videostab.so -lopencv_videostab -lASICamera2	
make: arm-bcm2708hardfp-linux-gnueabi-g++: Command not found
Makefile:28: recipe for target 'capture' failed
make: *** [capture] Error 127

anismo, 2017-11-28:
FYI: The install script has to be changed to copy the libASICamera2 libraries from the arvm6/armv7 subfolder as well. I noticed that with the latest pull, the shared libraries are soft links to libASICamera2.so.0.4.0929 however, that library itself is not included.

pi@allsky:~/allsky $ ls -lR lib
lib:
total 8
drwxr-xr-x 2 pi pi 4096 Nov 28 13:49 armv6
drwxr-xr-x 2 pi pi 4096 Nov 28 13:49 armv7

lib/armv6:
total 1168
-rwxr-xr-x 1 pi pi 1194600 Nov 28 13:49 libASICamera2.a
lrwxrwxrwx 1 pi pi      25 Nov 28 13:49 libASICamera2.so -> libASICamera2.so.0.4.0929

lib/armv7:
total 1036
-rwxr-xr-x 1 pi pi 1057600 Nov 28 13:49 libASICamera2.a
lrwxrwxrwx 1 pi pi      25 Nov 28 13:49 libASICamera2.so -> libASICamera2.so.0.4.0929

anismo, 2017-11-28:
Quick update:

Instead of the shared library (which is a softlink), I switched to the included static library for armv6 and compiled it and it loads up (Awsome!)

I hacked up the Makefile

1. In the Armv6 , the make file tries to look for arm-bcm2708hardfp-linux-gnueabi-g++ and this is not available (only arm-linux-gnueabihf-g++ is available). So I switch to using arm-linux-gnueabihf-g++

2. Instead of shared library -lASICamera2.so, I switch to static compile

capture:capture.cpp
        $(CC)  capture.cpp -o capture $(CFLAGS) $(OPENCV) lib/armv6/libASICamera2.a -lusb-1.0

Once I compiled it, the app started with no crash and started to take images with the ASI120MC USB 2.0 camera  (though it was blank but that could be my settings that I need to check now)

thomasjacquin, 2017-11-28:
Thanks for the debugging on your side.
I have updated the README and re-added the shared libraries.
I compiled using the armv6 and the armv7 flag without issues on my armv7-compatible Pi2.
I will close the issue for now. I'll re-open it if you come across any issues

closed 2017-11-28 by thomasjacquin

anismo, 2017-11-28:
Thank you so much! This is such an useful program. I got it working with my ASI224MC . I probably need to update firmware on my ASI120MC (it only gives black frames regardless of exposure).