AllskyTeam/allsky #10: Deletion of image directories from web GUI fails

mindpulse, 2018-02-25:
Dear Thomas,
I have been experiencing this issue across different raspbian releases and allsky updates: when trying to delete a directory from the web GUI, I get the notice "Deleted directory [date]" but the directory is not deleted. By inspecting the directories after the tentative deletion, I can see files belong to user "pi" and group "www-data" but they appear with permissions "-rw-r--r--", so the web server is unable to delete them. I am not sure how to troubleshoot this further.

uzelessknowledge, 2018-02-25:
I just reloaded AllSky using the latest code and it deleted for me. Might want to give that a try. 
although you have to backup and reconfigure somethings to update; small price.

thomasjacquin, 2018-02-26:
This [commit](https://github.com/thomasjacquin/allsky-portal/commit/d289117763c5e512a36a6f5a9bbcd9d073f3742a#diff-bb2b21ef7774df8687ff02b0284505c6) from January 22 fixes the issue. Did you clone the repository before that date?

mindpulse, 2018-02-26:
Yes, sorry, I just updated from scratch and all works fine now. Thanks!

closed 2018-02-26 by mindpulse