AllskyTeam/allsky #9: YouTube Upload?

uzelessknowledge, 2018-02-25:
Hey Thomas and gang. I've spent my whole weekend getting my all sky set up and have enjoyed the one night so far. 
I found this https://github.com/youtube/api-samples/tree/master/python. I was able to get it to upload the video, but you have to specify a name of the file and since the file is based on the date I couldn't figure out a workaround. Also has the ability to add a title, but could be the same name of the file, the date.
This guy also has a similar thing, but it ads the ability to specify what playlist you want to put it into. Still needs to specify a file name though. 

Maybe you guys can look into the ability to upload to YouTube or even vimeo?
Great work so far. Can't wait to get mine dialed in.

thomasjacquin, 2018-02-25:
Thank you for the link. That's an interesting feature to have, especially for users who don't have a hosting account.
I'll take a look at the code. In the meantime, if you want to call the timelapse differently, you can edit scripts/timelapse.sh line 22 and replace allsky-$1.mp4 by something else.
An easy workaround for you would be to remove $1 and call it simply allsky.mp4

uzelessknowledge, 2018-02-25:
Yeah. That’s why I started searching. I don’t have a hosting service. I may in the future but will just use YouTube for now. 
I’m content with the file names, but that does give me an idea. I’d either have to change the file name or title when uploading. 
Great project!

uzelessknowledge, 2018-02-26:
Anis has been helping me with my setup. I had it out last night for the first time. Tonight is all clouds and rain. 
I used the code from the Google API, upload_video.py. I call it using their method but I added it to the time-lapse.sh script. Anis pointed out how to pass the file name. I tired running timelapse.sh manually, but got a bunch of errors. 
mv: cannot stat './images/20180225/images/20180225/image-20180225195612.jpg': No such file or directory
However when I echo out this line I get the right file names. This will hopefully upload to Youtube with the title being the date, 20180225 for tomorrow, and a category of Science and Technology. 
python scripts/upload_video.py --file=images/$1/allsky-$1.mp4 --title=$1 --category="28"

This will at least get it there and then I will manually add it to the playlist. In theory. 

I'll let you know how this works. Fingers crossed. 

[upload_video.txt](https://github.com/thomasjacquin/allsky/files/1757005/upload_video.txt)

thomasjacquin, 2018-02-26:
Just fixed the timelapse script. You can change line 20 for the following code:
```
gawk 'BEGIN{ a=1 }{ printf "mv -v %s images/'$1'/%04d.'$EXTENSION'\n", $0, a++ }' |
```
I have pushed my changes to the repo as well.

uzelessknowledge, 2018-02-26:
Thanks Thomas. I update the script just in time for it to run. I got the correct timelapse built. 

The upload failed. I know when I was testing that I had to change directories into the directory where the upload script is. There is also a secrets file in this same location that has the api key. I may need to do a cd in the script to change into that directory to run. I’ll see when I can try that.

uzelessknowledge, 2018-02-26:
OK. I have the upload python script, the secrets file and the oath2 file all in the scripts directory. Added a CD to that directory and it runs and uploads. Also using the variable for the date, which creates the title to match the video. 

I can confirm that using googles sample script, unedited, will upload to YouTube. I'll just manually add the videos to the playlist, but I may play with toklands script and see what i can get to work. Hope this has been helpful. 
Now if there was some way to integrate this into the AllSky app. 

cd /home/pi/allsky/scripts/
python upload_video.py --file=/home/pi/allsky/images/$1/allsky-$1.mp4 --title=allsky-$1 --category="28" --keywords="AllSky Camera,All Sky Camera,Timelapse" --description="Timelapse over night with AllSky Camera"

thomasjacquin, 2018-02-27:
Sweet ! I'll take a look at the code and find the best way to integrate it without breaking anything.

uzelessknowledge, 2018-02-27:
Well, this morning everything kicked off as expected and my video I on YouTube, including a link to this github, but now I can’t access the pi. That happened yesterday too. Not sure why but after it finishes everything it’s no longer reachable. Doesn’t even ping. 
I’ve got a pi 3 coming tomorrow and maybe that’ll help. I did notice during the timelapse build it reaches 99% cpu utilization. So I don’t know if it’s that or the YouTube upload process. I may try adding a reboot to the timelapse script so when it finishes it reboots.

thomasjacquin, 2018-02-27:
Glad to hear the upload went fine. Not sure about the loss of access to the Pi. You may need to try to launch the script manually and see if it outputs anything in the console in the end.
Regarding cpu, avconv is a pretty intensive task but if you want to throttle it, you can uncomment line 21 on allsky.sh. It will add cpulimit to the avconv task.

uzelessknowledge, 2018-02-28:
It seems to have recovered after some time. Don't know when since I was at work, but later in the day I checked and was able to get back to it.

jcauthen78, 2018-02-28:
Glad to hear and see that the video uploaded nicely - hope the crash doesn't happen again for you tonight. I'm glad the text overlay worked well for you - and a good idea of sticking it it before the creation of the live-view, gonna borrow that idea.

Mixerboy24, 2021-07-16:
Hi,

Does this script still work?
And What does that json file contain? 

Can I just add a python script after it to get both Video to Youtube and the original directory?

Thanks.

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software and this issue appears to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-05 by EricClaeys