AllskyTeam/allsky #13: Host key verification failed

uzelessknowledge, 2018-03-09:
After the time-lapse builds this is the error I'm seeing. The file is created, but won't play. 
This is a fresh install on a Pi3. 
./timelapse.sh: line 22: 18769 Segmentation fault      avconv -y -f image2 -r 25 -i images/$1/%04d.$EXTENSION -vcodec libx264 -b:v 2000k -pix_fmt yuv420p images/$1/allsky-$1.mp4
cd: Fatal error: Host key verification failed.
put: images/20180308/allsky-20180308.mp4: Fatal error: Host key verification failed.

[timelapse.txt](https://github.com/thomasjacquin/allsky/files/1796896/timelapse.txt)

thomasjacquin, 2018-03-09:
I believe there are 2 different issues here:
Host key verification failed happens when it's the first time you're trying to upload to your server. lftp doesn't ask you if you're trusting this server and want to add the key to your known hosts in /home/pi/.ssh
In order to fix it, try to ssh to the server. It will prompt you to add the key: 
```
ssh <you_server_ip_address>
```

The segmentation fault comes from avconv. It could be a bad image in the stack of images or something else. Hard to say because avconv is not super chatty about the cause of the failure.
You can try running the timelapse command manually and see if it always fails at the same spot.

For what it's worth, this issue happened to me before when I was running the RPi inside my house. I believe avconv was making the cpu run close to 100% and the RPi was overheating. I added a line  (21) in allsky.sh which seemed to solve the issue for me.
```
cpulimit -e avconv -l 50 & \
```
Try uncommenting it and re-run the timelapse command. It will generate a bit slower but it should complete.
Note that I never had that issue with the RPi outside.

uzelessknowledge, 2018-03-10:
OK I fixed the ftp portion, but nothing I try will build a time-lapse that is usable. I've deleted a bunch of images, but they are all viewable. 
All I got for star trails was one image. It was the first clear night and not much of a moon. not complaining. Just anxious to see a full night. 

Again last night it failed. I'm creasing the SD and starting over. Something is corrupt. 

I ended up copying off the images and made my own time-lapse and star trails. SUPER COOL!

uzelessknowledge, 2018-03-11:
I formatted the SD card. Reloaded everything. Copied over the files from 20180309. Manually ran timelapse.sh. Timelapse, keogram, and star trails were built. Assemble everything and put it outside. Wakeup this morning and have no timelapse. I see a 14986 bus error. Tried to run it manually and have illegal instruction. 
./timelapse.sh: line 22: 18341 Illegal instruction     avconv -y -f image2 -r 25 -i images/$1/%04d.$EXTENSION -vcodec libx264 -b:v 2000k -pix_fmt yuv420p images/$1/allsky-$1.mp4

Guess I put the PI2 back out.

jcauthen78, 2018-03-11:
I've had my startrails come out as one single frame before (quite often), i've noticed it coming from mostly overcast/cloudy nights.. it seems the nights where its not too cloudy (some clouds here and there), the TL will work, but if it's mostly clouds with breaks to see the stars, it'll return just one image.  

I had a friend that lives down the road get random buss errors in the past doing video conversion, he thought it was an overheating problem, so he used heat syncs - it lessened the errors. he swapped out his memory card after that and the errors went away - not saying that's the problem, but could be something worth giving a shot.

uzelessknowledge, 2018-03-11:
Thanks for the input I already have heatsinks in place. I don't have another 64Gb card for the 3, but can try a 16Gb. 
The single image does happen more on cloudy nights, but also on a clear night. I changed line 32 in endofnight.sh where the 0.1 was to a 0.2 and it seemed to help some. 
I'm rebuilding the SD now for the pi2. I moved to a pi3 because of the speed, but seems to be more problematic.

thomasjacquin, 2018-03-11:
You can adjust the brightness threshold in endOfNight.sh line 32. It is set to 0.1. You could increase it to 0.2 if you live in a place with lots of light pollution and re-rerun the startrails command manually.

When the threshold is not reached, the image with the lowest brightness will be used as the result of the command.
As I mentioned earlier, I've had issues with avconv never finishing the job but that was only during a test inside, in summer without heatsinks so I can't tell whether the SD card, the heat or the Pi was the cause of failure.

uzelessknowledge, 2018-03-11:
Thanks Thomas. I've rebuilt it again and will see what happens tonight. I also have he pi2 stood up and ready to go. 
I had heat shrinks on the 3, but didn't on the 2. Took the larger one off to see it it will let it breather better. I may pick up another 64gb card.
More to come tomorrow. thanks guys!

uzelessknowledge, 2018-03-13:
Failed again this morning. Pullled it out of the case and inside. Changed it to 1 cpu and it ran fine. 2 cpu and time lapse was built. Removed maxcpus line and it built fine again. Go figure. Don’t think it was too warm since it was in the low 40’s this morning. 
Thanks for all your help.

closed 2018-03-13 by uzelessknowledge