AllskyTeam/allsky #15: Issues with FTP

dhavalbbhatt, 2018-04-14:
Hello,
I am having issues with FTP'ing the images to my website. It keeps giving me an error message indicating that there is a mismatch in the SSL certificate name. Anyone know how to suppress checking the SSL certificate name? I believe that is an OS level setting and not necessarily allsky setting? 

Thanks,
Dhaval

jcauthen78, 2018-04-15:
I had the same issue with my setup in the past - Thomas's suggestion worked for me. the manual terminal connection worked well, and then It popped up and asked me if i wanted to add the host cert (say yes) , since then, it's worked flawlessly through the script.

dhavalbbhatt, 2018-04-15:
Thanks guys! I will look at doing that. Another thing - is there a way to control when to start and stop the program (I realize that there is sunwait program that checks for civil twilight), but I would like to control when the program starts and ends taking images.

Thanks,
Dhaval

thomasjacquin, 2018-04-16:
Do you want to start/stop everyday at the same time? If so, you could rely on a CRON job.

Other option:
Sunwait's job is only to return "Day" or "Night" based on your GPS coordinates. You could write a script that controls when the capture should happen (it would return true/false or on/off or day/night, etc) and call that script in capture.cpp instead of calling sunwait. You'll have to re-compile capture.cpp after editing it.

thomasjacquin, 2018-10-07:
There is now an option to start capture based on the sun's position below the horizon.

closed 2018-10-07 by thomasjacquin