AllskyTeam/allsky #16: Black images

triplanedave, 2018-04-26:
Love the project, 100% fits the bill for what I'm trying to accomplish. Only challenge is that I'm getting pure black images. 

Log.txt is showing:

-bin 1 -brightness 50 -darkframe 0 - delay 20000 -exposure 5000000 -filename image.jpg -flip 0 -fontcolor 255 0 0-fontline 1 -fontname 0 -fontsize 0.7 -fonttype 0 -gain 50 - gamma 50 -height 960 -latitude 47.66N -longitude -122.12W -nodisplay 1 -quality 95 -text ThunderCreek -textx 15 -texty 35 -time 1 -type 1 -usb 80 -wbb 90 -wbr 53 -width 1280 

Thoughts on what I might need to change to get it properly imaging?  It's a ZWO120MC on a RPI 3 with the "compatible" firmware upgrade done on the camera.

triplanedave, 2018-04-26:
FWIW, setting back to default values makes no difference.  Using SharpCap with the camera, I am getting images, so not a broken camera.

thomasjacquin, 2018-04-26:
Hi Dave, did you try to capture images when it was dark outside?
The camera starts taking pictures at civil twilight so if you want to test it during the day, you'll have to set your lat/long to a place on earth where it's dark.
Let me know how it goes, I'm trying to narrow the issue down by eliminating the common problems.

bbillp, 2018-04-26:
That happened to me when I first used Allsky which I consider a nice feature, easier than adding a RTC and programming. Bill

triplanedave, 2018-04-26:
Yes, I left it running all night long.  It finally logged that it was out of daytime, and should be capturing images, but they were always black.

triplanedave, 2018-04-26:
Shouldn't live view in the Admin Panel at least show the current image, even if it's not capturing?

thomasjacquin, 2018-04-27:
Yes the live view shows image.jpg located at the root of the project. Have you tried starting the program manually? 
```
./allsky.sh
```
This may give you a bit more info in the terminal.

triplanedave, 2018-04-27:
No, hadn't thought of running it manually.  I'll give that a shot.  Also thinking about whether the metadata for the image might have some clue as to what the settings are for the capture.  Do you know if the metadata is provided by the camera driver, or the capture app, or ?  Would the image capture metatdata (exposure length, gain, etc) be of any value to know?

thomasjacquin, 2018-04-30:
I believe the only info being written to the metadata is image type, width
and height. If running the program manually still doesn't give you more
info, try going back to the old firmware.

thomasjacquin, 2018-05-03:
Hey Dave, you don't need to wait for night, you can change your longitude for some place on earth where it's dark in order to test the camera.
I have not seen this OpenCV error before. Could you try re-running the install.sh script and reboot in the end? I'm not sure why OpenCV would fail when trying to save the image.

bbillp, 2018-05-03:
Another method is to change the system date + dark hours with the command line date.  After your test then reboot and enter date to check that the date has returned to normal.    
Bill P.

triplanedave, 2018-05-05:
ok, doing the longitude works. manually running the script still errors some times, but sudo .\allsky.sh will eventually work.  But, still black images.
wondering if we can coordinate a remote session with teamviewer?  i have it running on the device.  maybe something will be obvious to you

triplanedave, 2018-05-06:
OK, problem solved.  
1) Reinstall Raspian
2) Update Raspian
3) Reinstall allsky
4) Use 2.4a power supply

All good!

thomasjacquin, 2018-05-06:
Great! The power supply may have been the issue here. The camera takes a bit of power so if the power supply is not strong enough, it will fail to work properly.

closed 2018-05-06 by thomasjacquin