AllskyTeam/allsky #18: Gui functionality broken

gshau, 2018-05-16:
The web-app doesn't appear to take any images.  I've changed lat/lon and the timezone to be in a simulated night environment, and there isn't any update to the live view page.  I can run the allsky.sh script just fine, and it updates the live view page.

gshau, 2018-05-16:
Thanks for the quick reply Thomas.  I’m editing the one on the webpage, which points to /var/www/html/.  

Gabe

thomasjacquin, 2018-05-16:
Does it update the image now? The capture should start automatically when allsky.sh starts after the Raspberry Pi has finished booting.
You can run `ps -aux` in the terminal to see what process is running. You should see `capture` in that list. Then you know it's capturing data.
If it didn't boot automatically, make sure you have a reference to allsky in `/home/pi/.config/lxsession/LXDE-pi/autostart`.
Also make sure you've chosen a real dark place on earth as the software only starts after civil twilight.

gshau, 2018-05-16:
Okay, thanks.  It’s odd, I had to create the .config/o session/LXDE-pi directory.  I’m running off raspian-lite.  Should `echo ‘@allsky’ >> ~/.config/lxsession/LXDE-pi/autostart` work?

thomasjacquin, 2018-05-16:
Oh ok, well in that case you probably don't have a graphical environment and you have to start allsky.sh manually. Creating the LXDE directory won't work because you don't have LXDE installed.

There's probably another way to start allsky.sh such as using systemd. I have rough instructions in this pull request: https://github.com/thomasjacquin/allsky/pull/14

That should work on any debian based distribution. Let me know if you get blocked.

gshau, 2018-05-16:
Thanks for the link.  I’ve tried it and had some progress.  It’s going through, but am met with:
`(video:744): Gtk-WARNING **: cannot open display:` instead.

gshau, 2018-05-16:
Doh!  Yes, I had that set incorrectly.  I'll close this issue now.  Thanks!

closed 2018-05-16 by gshau