AllskyTeam/allsky #19: Live view prob

jimjaxer, 2018-05-31:
Hello Thomas. Thanks for this great project.
I 'm experiencing a little pb with GUI interface. 
I've got no "Live view working" but a black image flickering sometimes and no image stored. No changes if I set different camera settings. I'm using an ASI 120MM with flashed firmware +RPI3
Checking for ACL in all directories but everything seems to be fine.
Regards
Raph

thomasjacquin, 2018-05-31:
Hi Raph,
Do you have a monitor attached or is it a headless setup?
If you have a monitor you should see the terminal with some debugging info. If it's currently dark at your place, it will start capturing images and output a line on the terminal "Saving...".

If you have a headless set up, you'll have to ssh to the RPi, kill a few processes (./capture, entr, allsky.sh) and start allsky.sh manually. Then you'll be able to see the debug info.

Keep in mind it's only saving new images during the night. You can play with the longitude in order to simulate the night at your place.

Thomas

jimjaxer, 2018-05-31:
Ok I'm with an ssh socket with MobaXterm this is a headless setup. Everything is fine with allsky.sh in a shell. 
Now it's fine a little reboot and original firmware of my 120MM of march 2017 works !
Trying with different ROI, full frame is too heavy for RPI3 in Live View excepted in wired network wifi is too slow.
Thanks !

closed 2018-05-31 by thomasjacquin