AllskyTeam/allsky #20: Want to upload to Dropbox? Use DropBox-Uploader by andreafabrizi

HPWever, 2018-06-02:
I like to have my files transferred to Dropbox rather than to an FTP server. I have therefore used the github project by andreafabrizi called DropBox-Uploader to enhance AllSky Camera. I have subsequently replaced all the LFTP commands in the "Endofnight.sh", "SaveImage.sh" and "Timelapse.sh" files with an easy: 
"/home/pi/Dropbox-Uploader/dropbox_uploader.sh upload ./allsky/images/$1/allsky-$1.mp4 /"
Works as a breeze.

topazkuntadi, 2020-12-23:
hi. i've already setup the allsky and dropbox uploader. i still don't understand wher do you put those
"/home/pi/Dropbox-Uploader/dropbox_uploader.sh upload ./allsky/images/$1/allsky-$1.mp4 /"

after lftp ? or just replace the whole line ?
what do $1 code for ?
and if i want to upload image, it would be $1.jpg ???
how about the other entry on the "SaveImage.sh"

Thanks a lot.

HPWever, 2020-12-23:
Hi,
They should be added to the Timelapse.sh script
Please check your directories (existence and name) carefully!
Below a section of my version (I commented out the lftp line):

if [ "$UPLOAD_VIDEO" = true ] ; then
# ftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$MP4DIR" -e "set net:max-retries 1; put images/$1/allsky-$1.mp4; bye"
# next line is added to upload to DROPBOX using the dropbox_uploader.sh
       /home/pi/Dropbox-Uploader/dropbox_uploader.sh upload /home/pi/allsky/images/$1/allsky-$1.mp4 /
fi

The S1 parameter should be automatically grabbed by the original ThomasJacquin scripts (it should get a value of todays date; so you will get a file called something like allsky-20201222.mp4)

Good luck and season greetings,
Harold

topazkuntadi, 2020-12-23:
Yes. It works. Thanks a lot.
Happy holiday & stay safe ...

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software and the issue appears to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-05 by EricClaeys