AllskyTeam/allsky #24: ASI290MM camera stops after a few hours

maphilli14, 2018-07-29:
OpenCV Error: Incorrect size of input array (Non-positive cols or rows) in cvInitMatHeader, file /build/opencv-U1UwfN/opencv-2.4.9.1+dfsg1/modules/core/src/array.cpp, line 146
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-U1UwfN/opencv-2.4.9.1+dfsg1/modules/core/src/array.cpp:146: error: (-201) Non-positive cols or rows in function cvInitMatHeader

/home/pi/allsky/allsky.sh: line 22:   956 Aborted                 ./capture $ARGUMENTS
cd `ftp://username@example.com:/allsky/' [Delaying before reconnect: 121]

thomasjacquin, 2018-07-30:
Hi,

Does this happen all the time? Is it during a normal night time capture or during timelapse generation?
Can you paste your configuration here?
Also, Can you set UPLOAD_IMG to false in config.sh? That will prevent the upload attempts.

Thanks

maphilli14, 2018-07-31:
`pi@allsky:~/allsky $ more config.sh
#!/bin/bash
source /home/pi/allsky/scripts/ftp-settings.sh

# Set to true to upload current image to your website
UPLOAD_IMG=false

# Set to true to upload timelapse video to your website at the end of each night
UPLOAD_VIDEO=false

# Set to true to send data to your server at the end of each night
POST_END_OF_NIGHT_DATA=false

# Set to true to generate a timelapse at the end of the night
TIMELAPSE=true

# Set to true to generate a keogram at the end of the night (image summary of th
e night)
KEOGRAM=true

# Set to true to upload the keogram after it has been generated
UPLOAD_KEOGRAM=false

# Set to true to generate a startrails image of the night. Will skip brighter im
ages to avoid over-exposure
STARTRAILS=true

# Set to true to upload the startrails after it has been generated
UPLOAD_STARTRAILS=false

# Set to true to enable automatic deletion of archived data (images + videos)
AUTO_DELETE=true

# Set this value to the number of archived nights you want to keep. Needs AUTO_D
ELETE=true to work
NIGHTS_TO_KEEP=14

# Path to the dark frame for hot pixels subtraction. Can be jpg or png.
DARK_FRAME="dark.png"

# Set to 1 to enable Daytime Capture
DAYTIME="0"

# Path to the camera settings (exposure, gain, delay, overlay, etc)
CAMERA_SETTINGS="/home/pi/allsky/settings.json"
pi@allsky:~/allsky $  
`

thomasjacquin, 2018-07-31:
That looks fine to me. Can you show me what you have for settings.json?

maphilli14, 2018-08-01:
Yes!  Thanks for the help!

```
pi@allsky:~/allsky $ cat settings.json
{  
   "width":"0",
   "height":"0",
   "exposure":"10000",
   "maxexposure":"10000",
   "autoexposure":"0",
   "gain":"0",
   "maxgain":"400",
   "autogain":"1",
   "gain":"50",
   "gamma":"50",
   "brightness":"50",
   "wbr":"53",
   "wbb":"90",
   "bin":"2",
   "delay":"10",
   "daytimeDelay": "5000",
   "type":"1",
   "quality":"95",
   "usb":"40",
   "filename":"image.jpg",
   "flip":"0",
   "text":"text",
   "textx":"15",
   "texty":"35",
   "fontname":"0",
   "fontcolor":"255 255 255",
   "fonttype":"0",
   "fontsize":"0.7",
   "fontline":"1",
   "latitude":"35.5N",
   "longitude":"78.5W",
   "time":"1",
   "darkframe":"0"
}
pi@allsky:~/allsky $ 
```

Also it seems to only take one photo now and stop?!

```
pi@allsky:~/allsky $ sudo service allsky status
* allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: e
   Active: active (running) since Tue 2018-07-31 21:59:55 EDT; 4min 18s ago
 Main PID: 771 (allsky.sh)
   CGroup: /system.slice/allsky.service
           |-771 /bin/bash /home/pi/allsky/allsky.sh
           `-851 ./capture -autoexposure 0 -autogain 1 -bin 2 -brightness 50 -da

Jul 31 21:59:55 allsky systemd[1]: Started All Sky Camera.
pi@allsky:~/allsky $ ls images/20180731/
image-20180731215529.jpg  image-20180731220051.jpg
image-20180731215740.jpg  thumbnails
pi@allsky:~/allsky $ ls images/20180731/
image-20180731215529.jpg  image-20180731220051.jpg
image-20180731215740.jpg  thumbnails
pi@allsky:~/allsky $ sudo service allsky status
* allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: e
   Active: active (running) since Tue 2018-07-31 21:59:55 EDT; 4min 46s ago
 Main PID: 771 (allsky.sh)
   CGroup: /system.slice/allsky.service
           |-771 /bin/bash /home/pi/allsky/allsky.sh
           `-851 ./capture -autoexposure 0 -autogain 1 -bin 2 -brightness 50 -da

Jul 31 21:59:55 allsky systemd[1]: Started All Sky Camera.
pi@allsky:~/allsky $ ls images/20180731/
image-20180731215529.jpg  image-20180731220051.jpg
image-20180731215740.jpg  thumbnails
pi@allsky:~/allsky $ 

```

maphilli14, 2018-08-01:
Also I swapped the 290 for a 174MM

maphilli14, 2018-08-01:
also, can I make a feature request?  Last 1min video loop?!  That's be great to spot really high, thin clouds.

thomasjacquin, 2018-08-01:
Was it dark when you tried the capture?
The way it works is that it will only take images from dusk to dawn based on your GPS coordinates. You can change them to see if it starts saving new images correctly.

maphilli14, 2018-08-02:
It was dark but fooling around I got it to take some photos but only when I stop start it...  also, why is this image from this AM in last night's folder?

```
Name↓ | Last Modified: | Size: | Type:
-- | -- | -- | --
../ |   | - | Directory
thumbnails/ | 2018-Aug-01 10:34:08 | - | Directory
image-20180731215529.jpg | 2018-Jul-31 21:55:29 | 267.6K | image/jpeg
image-20180731215740.jpg | 2018-Jul-31 21:57:40 | 109.5K | image/jpeg
image-20180731220051.jpg | 2018-Jul-31 22:00:51 | 110.0K | image/jpeg
image-20180731220836.jpg | 2018-Jul-31 22:08:36 | 104.6K | image/jpeg
image-20180731220904.jpg | 2018-Jul-31 22:09:04 | 106.2K | image/jpeg
image-20180731221006.jpg | 2018-Jul-31 22:10:06 | 106.2K | image/jpeg
image-20180731221328.jpg | 2018-Jul-31 22:13:28 | 101.9K | image/jpeg
image-20180731221651.jpg | 2018-Jul-31 22:16:51 | 9.6K | image/jpeg
image-20180731221817.jpg | 2018-Jul-31 22:18:17 | 99.9K | image/jpeg
image-20180731221842.jpg | 2018-Jul-31 22:18:42 | 101.9K | image/jpeg
image-20180731221907.jpg | 2018-Jul-31 22:19:07 | 108.6K | image/jpeg
image-20180731222324.jpg | 2018-Jul-31 22:23:24 | 155.7K | image/jpeg
image-20180731222404.jpg | 2018-Jul-31 22:24:04 | 154.5K | image/jpeg
image-20180731222453.jpg | 2018-Jul-31 22:24:53 | 107.3K | image/jpeg
image-20180731222547.jpg | 2018-Jul-31 22:25:47 | 90.9K | image/jpeg
image-20180731222718.jpg | 2018-Jul-31 22:27:18 | 12.0K | image/jpeg
image-20180731222728.jpg | 2018-Jul-31 22:27:28 | 12.1K | image/jpeg
image-20180731222735.jpg | 2018-Jul-31 22:27:35 | 12.1K | image/jpeg
image-20180731222745.jpg | 2018-Jul-31 22:27:45 | 12.0K | image/jpeg
image-20180731222840.jpg | 2018-Jul-31 22:28:40 | 90.1K | image/jpeg
image-20180731222911.jpg | 2018-Jul-31 22:29:11 | 89.5K | image/jpeg
image-20180731223215.jpg | 2018-Jul-31 22:32:15 | 155.4K | image/jpeg
image-20180731223433.jpg | 2018-Jul-31 22:34:33 | 153.4K | image/jpeg
image-20180731224748.jpg | 2018-Jul-31 22:47:48 | 151.5K | image/jpeg
image-20180731230138.jpg | 2018-Jul-31 23:01:38 | 161.8K | image/jpeg
image-20180731230322.jpg | 2018-Jul-31 23:03:22 | 159.9K | image/jpeg
image-20180731230706.jpg | 2018-Jul-31 23:07:06 | 121.4K | image/jpeg
image-20180801103407.jpg | 2018-Aug-01 10:34:07 | 184.9K | image/jpeg

```

Current settings file....

```

pi@allsky:~/allsky $ cat settings.json
{
   "width":"0",
   "height":"0",
   "exposure":"9000",
   "maxexposure":"9000",
   "autoexposure":"0",
   "gain":"150",
   "maxgain":"400",
   "autogain":"1",
   "gain":"150",
   "gamma":"50",
   "brightness":"50",
   "wbr":"53",
   "wbb":"90",
   "bin":"2",
   "delay":"10",
   "daytimeDelay": "5000",
   "type":"1",
   "quality":"95",
   "usb":"40",
   "filename":"image.jpg",
   "flip":"0",
   "text":"text",
   "textx":"15",
   "texty":"35",
   "fontname":"0",
   "fontcolor":"255 255 255",
   "fonttype":"0",
   "fontsize":"0.7",
   "fontline":"1",
   "latitude":"35.5N",
   "longitude":"78.5W",
   "time":"1",
   "darkframe":"0"
}

```

maphilli14, 2018-08-02:
[image]

maphilli14, 2018-08-02:
It seems to be working now, could it have to wait a whole day/night cycle before working?

maphilli14, 2018-08-02:
Found it!

[image]

maphilli14, 2018-08-02:
```

root@allsky:/home/pi/allsky# cat /var/log/allsky.log
Aug  1 21:34:06 allsky allsky.sh[14042]: convert-im6.q16: insufficient image data in file `image.jpg' @ error/jpeg.c/ReadJPEGImage/1039.
Aug  1 21:34:06 allsky allsky.sh[14042]: convert-im6.q16: no images defined `images/20180801/thumbnails/image-20180801213404.jpg' @ error/convert.c/ConvertImageCommand/3258.
Aug  1 21:34:06 allsky allsky.sh[14042]: /home/pi/allsky/allsky.sh: line 25: 14081 Segmentation fault      ./capture $ARGUMENTS
root@allsky:/home/pi/allsky#

```

maphilli14, 2018-08-02:
[image]

thomasjacquin, 2018-08-02:
Ok, thanks for tracking that down. Thumbnails generation is a new feature and it seems to fail when the source is empty or incomplete. I'll take a look at it. It should be fairly easy to fix.

Regarding your image showing in yesterday's directory, this is normal. The way directories are organised is the following: Make a new directory for any image taken after noon (12:00pm) and keep putting images in there until noon next day. This is considered a "Night".

maphilli14, 2018-08-02:
Sweet!  Once I'm done I'll have you add me to your maps and "I made this".  Would you consider having another section of the gui that gave a 'last X-minute' animation?  basically just take the last 60s and make a movie that overwrites itself to give a view of the high clouds I seem to have all too often

closed 2018-10-07 by thomasjacquin