AllskyTeam/allsky #27: Skipping images

alexdumitriu, 2018-08-10:
Hi Thomas,

I have noticed something weird while looking in the images saved during each night - it looks like my pi is not saving all the images - sometimes it's skipping a few minutes, other times hours. The odd thing is that the image-resize upload works fine (I looked in the log).

I take 15s images, at binning 2 and 100ms delay. What do you think? Should I increase the delay, or might it be a hardware issue (sd card?)

Here's a log extract:
`Saving image-20180808235446.jpg

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 
[... 585 lines cut ...]
Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Resizing image.jpg 

Uploading image-resize.jpg 

Saving image-20180809003543.jpg`

And the saved images:

`Saving image-20180808235446.jpg
Saving image-20180809000006.jpg
Saving image-20180809000542.jpg
Saving image-20180809000658.jpg
Saving image-20180809000729.jpg
Saving image-20180809000931.jpg
Saving image-20180809003411.jpg
Saving image-20180809003543.jpg`

Many thanks,
Alex

thomasjacquin, 2018-08-10:
You could try increasing the delay but I don't see why it would be the problem during night time.
Can you show me your settings.json file? What camera are you using?

alexdumitriu, 2018-08-11:
I'm using an ASI178MC with a RPI3. Here are the contents of settings.jason from the www/html folder (I'm using the web-gui):

`{"width":"0","height":"0","exposure":"15000","maxexposure":"30000","autoexposure":"0","gain":"150","maxgain":"200","autogain":"0","gamma":"50","brightness":"50","wbr":"53","wbb":"90","bin":"2","delay":"100","daytimeDelay":"10000","type":"1","quality":"100","usb":"40","filename":"image.jpg","flip":"0","text":"text","textx":"15","texty":"35","fontname":"0","fontcolor":"255 255 255","fonttype":"0","fontsize":"0.7","fontline":"1","latitude":"55.8N","longitude":"4.2W","time":"1","darkframe":"0"}`

thomasjacquin, 2018-08-11:
Can you list the first few images from that night?
```
ls *.jpg
```
I would like to see if it's an issue with the logging of messages or something else. Also, the sensor of the ASI178 is fairly large so it could be coming from a memory usage problem. What's the average size of an image in Kb or Mb?

alexdumitriu, 2018-08-14:
Hi Thomas,

Here's the result (synced over to a windows pc):
` Directory of C:\Users\Alex\Desktop\all\20180808

10/08/2018  13:42           407,156 0001.jpg
10/08/2018  13:33           410,366 0002.jpg
10/08/2018  13:33           414,343 0003.jpg
10/08/2018  13:38           417,579 0004.jpg
10/08/2018  13:41           420,454 0005.jpg
10/08/2018  13:39           423,120 0006.jpg
10/08/2018  13:42           425,702 0007.jpg
10/08/2018  13:42           428,035 0008.jpg
10/08/2018  13:37           431,088 0009.jpg
10/08/2018  13:39           432,850 0010.jpg`

I have played about with the delay, but it's still the same. I will probably back it up and start from scratch - maybe the memory card is giving up

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.
Version 0.8 decreases the chance for missed frames, and warns when the software tries to write an image when a prior write is still in progress.  Setting "Debug Level" to 3 displays the time it takes to write an image in /var/log/allsky.log, aiding in determining where the problem may be.

closed 2021-09-05 by EricClaeys