AllskyTeam/allsky #28: Dark Frames

triplanedave, 2018-08-10:
Question - when I take a dark frame, it's all white, with the timestamp on it. 

Is that what to expect?  Here's the file I just captured - 120MC camera has a cap on the lens, and is covered with 3 layers of towels. 

[image]

triplanedave, 2018-08-10:
And here's my config file.  
Do you have suggestions on gain/etc settings?  Attached is one of my recent pics, which just doesn't seem as good as it could be.

[settingsjson.txt](https://github.com/thomasjacquin/allsky/files/2278712/settingsjson.txt)
[image]

thomasjacquin, 2018-08-10:
Hi Dave,
No, this is not what you would expect from the dark frame. A successful dark frame would be all black, no timestamp and a few hot pixels here and there. There's definitely an issue here. Are you still on v0.5 or using the latest version (v0.6)?
The settings.json file should contain a few more options (auto exposuree, auto gain, etc) but it doesn't really matter for the problem you're seeing. I will create a new dark tonight and see if I experience the same issue.

triplanedave, 2018-08-10:
From the readme, the version history only goes to 0.5, so I presume I'm on v0.5.

thomasjacquin, 2018-08-11:
Maybe try to update to the latest version. I don't see anything wrong with your settings. At 6.5s and with a lens cap, it shouldn't be all white. Have you tried to bring gamma down?

triplanedave, 2018-08-12:
Updated to latest version, all is working now.  Thanks.

closed 2018-08-12 by triplanedave