AllskyTeam/allsky #29: How to change time/date color?

Mixerboy24, 2018-08-11:
Hi Thomas.

How do I change the color of day / time from blue to orange?
How do I get the text as well? settings.json in the text will read "Mixerboy24Tech Allsky" but it will not read in the picture. I'm sorry for my bad english :(

Atte / Mixer

thomasjacquin, 2018-08-13:
It is possible by modifying capture.cpp and recompile the program by running `make capture`.
If you feel confident writing a few lines of C++, then you should be able to add it. 
After line 550, you can add something like 
`cvText(pRgb, bufTime, YourXposition, YourYPosition, fontsize, linewidth, linetype[linenumber], fontname[fontnumber], fontcolor, Image_type);
`
Just select a value for YourXposition and YourYPosition
Then you can recompile and restart the service.

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software it's now possible to change the color using the GUI.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys