AllskyTeam/allsky #30: Live View refresh rate

astro-mb, 2018-08-11:
Hi Thomas,

First of all, big thanks for this software ! Very usefull for my needs, thanks again.

The Live view refresh rate use the exposure time to refresh itself, I think it would be better to use the delay between images to refresh the live view (half of the time would be better actually).

In index.php : 
 `setInterval(function(){
                        getImage();
                }, <?php echo $camera_settings_array["exposure"] ?>);`

changed to : 
 `setInterval(function(){
                        getImage();
                }, <?php echo $camera_settings_array["delay"]/2 ?>);`

Cheers, 👍 

Matt

thomasjacquin, 2018-08-11:
Hi Matt,

The reason I used exposure was to make sure I would only refresh once per new image. For example, if you have a 30s exposure and 1 sec delay between images, the UI would fetch the new image every 30 seconds. If I used delay, I would refresh 30 times and still display the same image. That has an impact on bandwidth and can be problematic for some who have limited internet access.

But I understand your point here, by using `exposure`, the "live" view is not really live, it's always a bit in the past.

Actually, during daytime, It would make sense to use `daytimedelay + exposure` and at night `delay + exposure` in order to refresh at a reasonable pace.

astro-mb, 2018-08-12:
Hi Thomas,

OK, thanks for your answer.

Matt

closed 2018-08-12 by astro-mb