AllskyTeam/allsky #31: Images is completely black

wannaberocker, 2018-08-12:
Hello, I seem to be getting only a black image when I run the ./allsky.sh preview command. Also is there a way to start the camera with the GUI? What am I doing wrong? Thanks. Using a ZWO ASI178 mono.

thomasjacquin, 2018-08-13:
Hi,
Can you copy what's in the terminal? That will help me track the problem down. If you install the GUI (sudo gui/install.sh) then you should be able to see the interface by going to http://localhost on the Pi or http://you-rpi-ip from any computer on your home network.
Thomas

wannaberocker, 2018-08-13:
I figured it out I think. I had to changes save daytime files to "1". Apperently it must have to save them to display them. Pretty cool stuff here.

thomasjacquin, 2018-08-13:
Initially, the program was only capturing images before dusk and dawn but in version 0.6, I added daytime capture to monitor the sky during the day. If you're trying the software during daytime and you have daytime set to 0, it won't save any images indeed.
Note that day time images are not saved (just overwritten) contrary to night time images which are stored in directories such as 20180812 for example.