AllskyTeam/allsky #32: Problem with day time images

simoleso, 2018-08-13:
Hi Thomas,
sometime, during day time, the image seems to be scrambled (i.e.: https://www.dropbox.com/s/r5mzjbjqhigx09h/allsky_error.jpg?dl=0). I've enabled auto exposure and the auto gain is disabled. I got an ASI120MC (USB2) and the last git version of the software.

After a few frames the return correct.

Thanks in advance for you attention,

Simone

thomasjacquin, 2018-08-13:
Hi Simone,

It sometimes happens to me as well during day or night time. It's purely a "preview" problem (saved images are fine).
I believe the reason is that the image displayed in your browser doesn't have enough time to load before its source is overwritten by saveImageDay.sh.
I think a slow internet connection + short daytimedelay is the issue here. This could be verified by increasing the delay between images (10 000ms for example). 
If you come across a broken image, try refreshing the browser to see if you get the full image.

A potential fix would be to use rotating temporary images but I haven't given too much tought into the implementation yet.

Thomas

uzelessknowledge, 2018-08-19:
Hey Thomas. I see you've added the ability to capture day time images, but they aren't saved. I've not touched my camera since April, and can't remember half of what I did, but was wondering if you have thought about saving them to generate a 24 hr time-lapse?

uzelessknowledge, 2018-08-20:
I added the save code to the day time .sh. I have it saving day images now. My new question is if auto exposure can be used in the day, but a defined exposure for night. I have not tired, but wondering if auto exposure will be long enough for the sky. I have some trees with light pollution and worried it will adjust to them.
My replay for night is based on the longer exposure value. Having images more often cause more images of course.

thomasjacquin, 2018-08-21:
The way it works now is:

- Day:
    - auto-exposure: always on
    - gain: 1
- Night:
    - exposure value OR auto-exposure with max exposure
    - gain value OR auto-gain with max gain

To keep a regular frame rate for the timelapse, when using auto-exposure, the delay between frames is equal to max-exposure value.

ex 1: No auto-exposure, 20sec exposure
 A new image is taken every 20sec (+ delay)

ex 2: auto-exposure, max-exposure 30sec
 A new image is taken (5sec), wait 25sec (+ delay)
 A new image is taken (7sec), wait 23 sec (+ delay)
 A new image is taken (9 sec), wait 21 sec (+ delay)
...
A new image is taken (30sec), wait 0 sec (+delay)

That way, we keep a constant frame rate and all the images are properly exposed. The drawback is that when you wait, you're not capturing the sky.

To answer your question, yes, it is possible to do 24hr timelapse. you just need to save the images in saveImageDay.sh. However, there is a bit of code I need to add to capture.cpp to start the night non-over-exposed.
Also, a new day starts at noon and ends at noon so, depending on which kind of "24hrs" you want, you may need to stitch 2 consecutive videos together.

uzelessknowledge, 2018-08-22:
Attached is what I added to saveImageDay.sh to get it to save the images. 

Here are my current settings. 
Auto-Exposure No, Exposure 8000, Max Exposure 10000, Delay 2000, Daytime Delay 30000
I'm getting images every 8-10 seconds

Then I tried Auto-Exposure Yes, Exposure 8000, Max Exposure 10000, Delay 2000, Daytime Delay 30000
I'm getting images every ~17 seconds

I have not had the chance to do daytime. If I need to open another issue to discuss let me know. With there was a discussion option and not have to open an issue. 

[log.txt](https://github.com/thomasjacquin/allsky/files/2308527/log.txt)
[saveImageNight.txt](https://github.com/thomasjacquin/allsky/files/2308528/saveImageNight.txt)
[saveImageDay.txt](https://github.com/thomasjacquin/allsky/files/2308341/saveImageDay.txt)

topazkuntadi, 2020-12-26:
i'm using zwo 290mm , nightime is running flawlessly. but not daytime.
after "service allsky restart", the software save some image, like 5-8 image. than freeze.
no more image saved. i feel it worst when the sun shine brighter.
has any one experience this ???

what log / SS I can sent here to debug / find the problem ?
Thanks.

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.
Version 0.8 of AllSky decreases the chance this will happen by performing many of the conversion tasks on a copy of the image so it's less likely a web page will access an image as it's being updated.

closed 2021-09-05 by EricClaeys