AllskyTeam/allsky #33: Timelapse isn't being generated

wannaberocker, 2018-08-18:
Is there a reason why the time lapse isn't being generated? The clip graphic shows up when choose time lapse and all the images have been saved but the time lapse won't play back. Do I need to have Save daytime images set to "0" fpr this to work? The first night I used this the time lapse was generated but not the second time I used this.

Thanks,

thomasjacquin, 2018-08-18:
The ASI178 has a very large sensor which produces large images (probably a few MB). The RPi is a very capable computer but it has its limits. Avconv is probably hitting a memory limit threshold and stops.
To make sure that's the issue, you can try running the timelapse manually:
`scripts/timelapse.sh 20180817`
Look for any errors, warnings, segmentation fault, etc and report them here.

My guess is that it will not always fail at the same image, which means there's no problem with the images themselves, just that the RPi has reached some memory limit.

One way to get things to work is to reduce the images
in scripts/timelapse.sh, on line 24, you can add something like that:
`magick mogrify -resize 50% images/$1/*.$EXTENSION`

That should reduce the length and width by 50% (image 4 times smaller on disk). Keep in mind that this command is overwriting the source images though.

wannaberocker, 2018-08-19:
I am a bit confused exactly here to put the: magick mogrify -resize 50% images/$1/*.$EXTENSION 

I don't know how to tell if I am on line 24 or at the right spot or if I am putting it in right

thomasjacquin, 2018-08-19:
Hey, I could have explained that better, I agree.
Here's what I would do:
```
# Shrink images coming from large sensors (ex: ASI178MM and ASI178MC). Adjust percentage to your own preference
magick mogrify -resize 50% images/$1/*.$EXTENSION

# Find invalid images (0 bytes size) and delete them to prevent errors during timelapse
find "images/$1" -name "*.$EXTENSION" -size 0 -delete

# find images, rename images sequentially and start avconv to build mp4; upload mp4 and move directory
ls -rt images/$1/*.$EXTENSION |
gawk 'BEGIN{ a=1 }{ printf "mv -v %s images/'$1'/%04d.'$EXTENSION'\n", $0, a++ }' |
bash
avconv -y -f image2 -r 25 -i images/$1/%04d.$EXTENSION -vcodec libx264 -b:v 2000k -pix_fmt yuv420p images/$1/allsky-$1.mp4
```
So I basically just added the mogrify command to reduce images size. I haven't tested it but I will try it with last night's data. If you want to be on the safe side, you can make a backup copy of last night and run timelapse manually. If 50% was not enough, you can restore your backup and try a lower percentage until the timelapse is successful.

Please keep me posted and I'll drop a note in the README for large sensors.

Thomas

wannaberocker, 2018-08-21:
I am a total newbie at this, but I seem to be unsuccessful so far, there was something about unable to encode the video...??. So then I tried it with a reduced number of images ....
( 40 instead of nearly 700, these are about 9.2 MiB in size so they are huge...) the script will report "Time lapse was created" but the mp4 is unplayable and the actual file in the pi is reporting the mp4 files is (0 bytes) and will not play. 

Oddly enough I did have a successful time lapse created the first time I tried this out but I have not since.

thomasjacquin, 2018-08-21:
Can you try running this directly from the terminal (adapt the date if you need to):
```
magick mogrify -resize 30% images/201808/17/*.jpg
```
And then look at the size of an individual image. It should be way less than 9.2MB.

Also, since your images are so large, have you checked how much space is left on your SD card?

wannaberocker, 2018-08-22:
Ok is is installed like you said. However I have a new issue....I am only getting black images. It was fine earlier when I was testing things.

wannaberocker, 2018-08-22:
No luck at all. What camera are you using? I am just going to buy the same one you use. The ZWO ASI178color cam does not work with this.

wannaberocker, 2018-08-23:
I added the text to the script as suggested above and when I run scripts/timelapse.sh 20180821 i get an error that looks like this:

[swscaler @ 0x1481300] depreciated pixel format used, make sure you did set the range correctly

and

x264 [error]: malloc of size 20515824 failed
video encoding failed
[libx264 @ 0x1469e00] final ratefactor: 30.79
Conversion Failed!
* Timelapse was created

The timelapse is not playable, has 0 bytes and is basically an empty file.

Also when I try to run magick mogrify -resize 30% images/201808/17/*.jpg I get an error that says:

bash: magick: command not found

What is the next course of action I should take? Turns out the saved .jpg are 254.1 KiB each approx.

thomasjacquin, 2018-08-23:
Sorry, I should have tested that first. You don't need the `magick` keyword. Try this instead:
```
mogrify -resize 30% images/201808/17/*.jpg
```
This should work and resize the images to something that avconv can handle.

wannaberocker, 2018-09-25:
I have tried a few things but from what I conclude it will try to make the timelapse but it always ends up as a "0 KiB" size file that doesn't play. ANy other bits of advic I can follow to track down the issue? It seems most files generated are now about 450 KiB after setting to JPG 80% quality. Thanks.

wannaberocker, 2018-09-25:
Yes when I run it manually usingn the script......it seems t generate an empty file. I have not tried to copy it and play it on my home PC, I just figured it would be an empty file. I'll try later when I get home.

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

Please see the README.me file on the GitHub main page for workarounds to the "timelapse not created issue".

closed 2021-09-05 by EricClaeys