AllskyTeam/allsky #37: rtsp stream

MekyHum, 2018-09-16:
Hello, I have a zwo178 and I just installing this on RPI 3B+, I have one question... can You make a rtsp live stream from the camera, I would like to stream a live video to second rpi and with the second pi do meteor detection, I need 25 fps ....

Thank You 
Aleksandar

thomasjacquin, 2018-09-16:
Hi,
I am not sure the raspberry pi could handle such a fast frame rate. The Ports are USB2 and it also depends on the SD card write speed. You can try setting delay to 0 and have the exposure equal to 1/25 s. That should create 25 new image every seconds.

MekyHum, 2018-09-17:
ok, thank you, i have a i5 with 8gb ddr3, and linux mint 64 bit on it, that will be ok for this ... but I need someone who can help me to do a rtsp stream from the camera.... anyone have some idea ?

ivanemme, 2018-11-07:
hi all, i need same issue. If thomas can write opencv image to /dev/video0 with v4l2loopback we can use vlc o gstream to get from /dev/video0 and stream

maphilli14, 2019-03-13:
This would be a neat feature!

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

@ivanemme, if you'd like to create this capability we would consider adding it to AllSky.  Our plate is quite full right now with other enhancements.

closed 2021-09-05 by EricClaeys