AllskyTeam/allsky #38: FTP Cert Verification question

jcauthen78, 2018-09-20:
I've recently upgraded to a new RPI 3 b+ and have it up and running for the most part, but I can't get the images to upload to my website. I know the username and password are correct, I can connect through FileZilla on my desktop. But when I manually try to connect through the RPI terminal window, I get an error: (i've redacted user & host info) 

```
pi@allsky:~ $ lftp ftp.myhost.com
lftp ftp.jimcauthen.com:~> user myuser@myhost.com
Password: 
lftp myuser@myhost.com@ftp.myhost.com:~> ls
ls: Fatal error: Certificate verification: certificate common name doesn't match requested host name ‘ftp.myhost.com’ (9D:31:D7:C1:F4:D1:3D:71:77:B6:EE:45:18:5E:1B:71:33:40:6F:97)
lftp myuser@myhost.com@ftp.myhost.com:~> 
```

the "log.txt" file shows the attempt of uploading, but no success/fail message.
```
20000000
Saving image-20180919233720.jpg

Resizing image.jpg 

Uploading image-resize.jpg 
```

I've verified the file on the host, and it's an old file from my previous AllSky. 

Any suggestions/troubleshooting steps would be greatly appreciated.

And as before, Thank You for this awesome project!

thomasjacquin, 2018-09-21:
Hi Jim,
lftp doesn't add the host to the "trusted hosts" by default.
Try connecting to your host via SSH:
`ssh myhost.com`
This should prompt you if you want to add this host to the list of trusted hosts. Just say yes and it should work after that.

jcauthen78, 2018-09-21:
I’ll give that a shot next week - thought I tried that after I read another thread, but not sure. 

Next week, because forgot to fully seal my case last night and, we’ll, the dew set in and fried the Pi. >.<‘

spacemanbobby, 2018-12-22:
I tried the ssh myhost.com but i think that is for port 22. When I go back and try lftp myhost.com and log in with the user creds I still get the dreaded Fatal error when trying to look around for directory or folders.

spacemanbobby, 2018-12-26:
> When I get in I try the command within lftp set ftp:ssl-allow no and then I can cruise around.. I am wondering if the FTP operation will be able to get in since I had to enter this command each time I manually connected via command line.

Images still not uploading to ftp site via the script. Account settings are the same that I tested on lftp in command line.

spacemanbobby, 2018-12-27:
> Have you tried updating the lftp calls in saveImageDay.sh and
> saveImageNight.sh?
> Change this:
> lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set
> net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION;
> bye" &
> into this:
> lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set
> net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put
> $FILENAME-resize.$EXTENSION; bye" &

Changing that now shows the image being put on my ftp site.  I just changed
the lftp settings on all the other scripts that had lftp and inserted this
after the time out value set ftp:ssl-allow no;

Tomorrow I'll see if it uploads the keogram startrails and video.

spacemanbobby, 2018-12-30:
Everything uploaded last night. Thanks for the help.

fabiopignata, 2023-11-21:
Hi Thomas, I have the same problem but can't edit saveImageDay.sh, apparently now all lftp instructions are in lftp_cmds.txt
How shall I edit that?
many thanks! Fabio

EricClaeys, 2023-11-21:
@fabiopignata, you shouldn't edit any Allsky programs.  A lot has changed in the last 5 years.  If you have the latest Allsky release, please see the documentation available from the WebUI. If you are having problems uploading files see the `Troubleshooting -> Uploads` page.