AllskyTeam/allsky #39: contrast

MekyHum, 2018-09-22:
Hello,
I am playing with asi174 camera and rpi 3B+, and I tested the system last night, it was moon but I put 10 seconds of exp, and gain to 280, gama 50-70, brightnes 70-90, but I have very dark images, I have a fujinon lens, and when I make time lapses with windows I see crealy milky...but here is to dark... I see that in confing missing contrast option ? 

Here is a link what I can see on windows with original software with thase lenses cca 10 sec exp.
https://www.youtube.com/watch?v=1OTL5P_Q5tM

And here is a image from last night with 13 sec

[image]

What I do wrong ?
Thank You
Aleksandar

thomasjacquin, 2018-09-23:
Hi Aleksandar, 
You're right, it looks a bit too dark for these settings. Can you send me the settings you're using? (screenshot of the GUI or copy of /var/www/html/settings.json)
I'll take a look at it.
Thomas

MekyHum, 2018-09-23:
{
   "width":"0",
   "height":"0",
   "exposure":"10000",
   "maxexposure":"10000",
   "autoexposure":"0",
   "gain":"270",
   "maxgain":"280",
   "autogain":"0",
   "gamma":"50",
   "brightness":"95",
   "wbr":"50",
   "wbb":"50",
   "bin":"1",
   "delay":"1500",
   "daytimeDelay": "10000",
   "type":"1",
   "quality":"95",
   "usb":"40",
   "filename":"image.jpg",
   "flip":"0",
   "text":"text",
   "textx":"15",
   "texty":"30",
   "fontname":"0",
   "fontcolor":"255 255 255",
   "fonttype":"0",
   "fontsize":"0.7",
   "fontline":"1",
   "latitude":"45.3N",
   "longitude":"14.04W",
   "angle":"0",
   "time":"1",
   "darkframe":"0",
   "showDetails":"1"
}

thomasjacquin, 2018-09-23:
Your settings seem alright to me.
Do you have a comparison between the RPi3 and Windows for the same night with the moon?
The ZWO libraries don't have a contrast option but I'm sure we'll figure out what's going on here.
Thomas

MekyHum, 2018-09-23:
I don't have but i will make comparision, if the option is not awelible for contrast, wich value of contrast is take ?

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.
ZWO cameras don't have an option for "contrast" although they do have "brightness" which may suffice.

closed 2021-09-05 by EricClaeys