AllskyTeam/allsky #41: Dark frame only generated at night

amiles347, 2018-09-25:
I couldn't understand why I couldn't generate a dark.png as per the instructions.  Until I realised that the dark frame code is only in saveImageNight.sh and I was setting up in the day!

thomasjacquin, 2018-09-25:
Oh yeah, sorry about that. There was no "day time" capture in v0.5 and I haven't had a chance to add the same code to `saveImageDay.sh`.

dmnkhhn, 2019-06-04:
Hello! I have compared both files and the dark.png generation is still missing from the `saveImageDay.sh` - is it enough to just copy the lines of code to the file? If so I will send you a pull request. :-)

thomasjacquin, 2019-06-04:
Hi, 

Yes it should work but I think you'll want to use the night time capture settings for this (daytime always uses gain 0).
This is what you probably want to do in capture.cpp:
```
[...]
if(dayOrNight == "DAY")
{
    // Setup the daytime capture parameters
    if(endOfNight==true)
    {
        system("scripts/endOfNight.sh &");
        endOfNight=false;
    }
    if(daytimeCapture!=1)
    {
        needCapture=false;
        printf("It's daytime... we're not saving images\n");
        usleep(daytimeDelay*1000);
    }
    else
    {
        if (darkframe == 1 ) 
        {
            // Set exposure value for darkframe capture (similar to night time setting in order to match gain and exposure)
            ASISetControlValue(CamNum, ASI_EXPOSURE, currentExposure, asiAutoExposure == 1 ? ASI_TRUE : ASI_FALSE);
            ASISetControlValue(CamNum, ASI_GAIN, asiGain, asiAutoGain == 1 ? ASI_TRUE : ASI_FALSE);
        }
        else
        {
            printf("Starting daytime capture\n");
            printf("Saving auto exposed images every %d ms\n\n",daytimeDelay);
            exp_ms=32;
            useDelay=daytimeDelay;
            captureTimeout=exp_ms<=100?200:exp_ms*2;
            ASISetControlValue(CamNum, ASI_EXPOSURE, exp_ms, ASI_TRUE);
            ASISetControlValue(CamNum, ASI_GAIN, 0, ASI_FALSE);
        }
        
    }
}
else if(dayOrNight=="NIGHT")
{
    // Setup the night time capture parameters
    if(asiAutoExposure == 1)
    {
        printf("Saving auto exposed images every %d ms\n\n", delay);
    }
    else
    {
        printf("Saving %ds exposure images every %d ms\n\n", (int)round(currentExposure/1000000), delay);
    }
    // Set exposure value for night time capture
    ASISetControlValue(CamNum, ASI_EXPOSURE, currentExposure, asiAutoExposure == 1 ? ASI_TRUE : ASI_FALSE);
    ASISetControlValue(CamNum, ASI_GAIN, asiGain, asiAutoGain == 1 ? ASI_TRUE : ASI_FALSE);
}
[...]
```

I haven't tested it but it basically force the use of night time settings for day time.

dmnkhhn, 2019-06-06:
Alright, I will take a look at it. Thanks! 👍

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software and has been resolved in newer versions of AllSky.

closed 2021-09-05 by EricClaeys