AllskyTeam/allsky #43: Camera settings not updating

rankinstudio, 2018-10-06:
Hey man, love the new features. 

I installed the GUI option. Everything seems to be working, but the camera won't actually change exposure settings.

No matter what settings I choose, the gain is always 0 and the exposure is always .064ms. 

Any ideas on this?

Thanks!

thomasjacquin, 2018-10-06:
Hey David,
What do you have in config.json for the variable CAMERA_SETTINGS?
I should be pointing at /var/www/html/settings.json
Can you also show me the content of /var/www/html/settings.json or just a screenshot of the settings page on the GUI.
Thanks

rankinstudio, 2018-10-06:
Hey Thomas,

The CAMERA_SETTINGS is pointing at that file

I went through and verified that when I make changes in the GUI the /var/www/html/settings.json is updating correctly. For some reason these changes just don't seem to be actually affecting the camera in any way.

/var/www/html/settings.json contents:

{"width":"0","height":"0","exposure":"10000","maxexposure":"20000","autoexposure":"1","gain":"50","maxgain":"200","autogain":"1","gamma":"50","brightness":"50","wbr":"53","wbb":"90","bin":"1","delay":"3000","daytimeDelay":"1000","type":"1","quality":"95","usb":"40","filename":"rankin.jpg","flip":"0","text":"","textx":"15","texty":"35","fontname":"0","fontcolor":"255 255 255","fonttype":"0","fontsize":"1","fontline":"1","latitude":"37.0N","longitude":"111.66W","angle":"-6","time":"1","darkframe":"0","showDetails":"1"}

thomasjacquin, 2018-10-06:
Can you try to reboot the Pi?
Alternatively, can you try `sudo service allsky restart` ?
Also, just to be sure, make sure that the allsky entry got deleted in `/home/pi/.config/lxsession/LXDE-pi/autostart`

thomasjacquin, 2018-10-06:
I see you have autoexposure and autogain set both to 1. For some reason, the ZWO library doesn't seem to handle it very well and throws random results (black image sometimes but not always).

rankinstudio, 2018-10-07:
The issues seem to only happen in daytime mode. I'll have to wait until tomorrow to troubleshoot more. Thanks Thomas!

thomasjacquin, 2018-10-07:
Ha! Let me save you some debugging trouble here: Daytime is set to always use autoexposure (deals with clouds, sun, shade, sunrise, sunset better than fixed exposure) and gain=0.
When night time begins, it uses the settings you specified in the Camera settings tab of the GUI.
Night times is controlled by the angle of the sun below the horizon (default -6 degrees). You can change it to start capturing earlier or later.

rankinstudio, 2018-10-07:
Hey Thomas,
The issue is that my daytime images are under-exposing, especially with cloud cover. Is there a way to get the daytime gain higher?

rankinstudio, 2018-10-07:
Also looks like this ASi120MM is still producing scrambled images on short exposures. No work around found for this yet either? Been an issue for years. 

http://www.rankinstudio.com/dnloads/Scamble.JPG

thomasjacquin, 2018-10-07:
For the gain, you could edit line 651 in capture.cpp and replace 0 by something else... 50 for example.
Save the file and run `make capture` to recompile the binary and `sudo service allsky restart` to restart the service.

The problem with scrambled images is on ZWO's side. They seem to have issues with USB 2 cameras. I think that's why they don't support them with ASI Air.
Did you upgrade the camera firmware?

rankinstudio, 2018-10-07:
Okay, thanks on the gain tip. I'll give that a shot. 

I did update to the latest compatible firmware, no changes. I guess I'll just get a 224 Color.

Cheers!

rankinstudio, 2018-10-07:
Fireballs I've caught since launching this thing: http://www.rankinstudio.com/Fireball_Detections

thomasjacquin, 2018-10-07:
Wow, that one from November 11th of last year was a pretty big one.

rankinstudio, 2018-10-07:
8 sec with the stock lens. 

Just hosed my ASI120MM usb2 after attempting another FW update. Ordered the ASI120MC-S.

rankinstudio, 2018-10-12:
Yeah, I don't get it. Just got a brand new ASI120MC-S. Fresh install of allsky. Daytime images are still coming in way under exposed.

rankinstudio, 2018-10-12:
It starts out over exposed, then I can see it bracket down until it thinks it's okay? The exposure is way too dark though.

rankinstudio, 2018-10-12:
This is also strange. I had auto exposure enabled when it switched to nighttime mode, and it start to auto expose correctly. Must be a bug with the daytime mode auto exposure code?

scubabri, 2018-10-12:
Most of the time I have to reboot after nightfall to get auto exposure to correct for night time otherwise all the images are black. Once I reboot, it takes a few subs to get a correct exposure and then its good until the next night.

[cid:image001.png@01D4619A.2C0A08E0]

Somewhat annoying if I forget.

thomasjacquin, 2018-10-12:
Hey David, can you post a picture of daytime for reference?
Also, are you using autoexposure as well as autogain? Theses 2 options don't seem to play together very well. I assume this is an issue on the ZWO libraries regarding which option gets precedence.
I will review the daytime loop code to see if there could be a bug in it. FYI, on my allsky, daytime exposure seems correct; a bit dark when it's cloudy but properly exposed on blue sky.
Thomas

thomasjacquin, 2018-10-12:
Brian,
I experienced the same thing when I used auto-exposure and auto-gain at the same time. Do you have both enabled by any chance? On mine, I don't use auto gain, I just set the gain somewhere around 250 and enable auto-exposure. It gives good results from dawn to dusk.
Thomas

rankinstudio, 2018-10-12:
Thomas, when I say under exposed, I mean almost black. It is completely under exposed. It was that way just a few minutes ago. I changed the settings to 50 deg above horizon for recording to force capture during daytime (put it into nighttime mode) and it works perfectly with auto exposure on, and auto gain off. http://rankinstudio.com/dnloads/morning.JPG. There has to be a bug. When night mode is off, the camera only takes .0064ms exposures at gain 0. It doesn't change the exposure time at all.

rankinstudio, 2018-10-17:
Okay, I think this is resolved. I had auto exposure off, assuming it would kick on during the day? It seems if I just leave auto exposure on, things are okay. Cheers!

thomasjacquin, 2018-10-17:
Glad to hear that.
It SHOULD kick off during the day regardless of the auto-exposure setting. If it doesn't then it's a bug. I'll take a look at it. Thanks for reporting it.

closed 2018-10-17 by thomasjacquin