AllskyTeam/allsky #46: Connection Status

theoml, 2018-12-02:
Hi !

Just did the Project and it´s working fine. Thanks a lot for this!

In connection status I do not see any Parameter in Interface Information and Statistics. Just wlan 0

Is this normal ?
I use the new PI A+, (which is not recognized at system) und did a fresh setup with a PI 3+. Here it´s the same. 

Any idear ?

Regards from Germany
Reiner

thomasjacquin, 2018-12-03:
Thanks for reporting the issue.

I believe this is related to this issue: https://github.com/thomasjacquin/allsky-portal/pull/1/commits/20c58a2b9f0d4721171b92681770e765ee6d1df8

Basically ip & iw are now used instead of ifconfig & iwconfig. I will test it, make sure it still works for RPi3 and under and I'll push an update.

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issue appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys