AllskyTeam/allsky #47: Network Issue

drt102, 2018-12-13:
My system doesn't record any images when it isn't connected to a network. Is this a design feature? If so, is there a way that this behaviour can be altered as I would like to be able to take the allsky camera to a dark sky site where there is no network connectivity. I'm using the allsky setup with the portal.

thomasjacquin, 2018-12-14:
The system should take images whether the system has a network connection or not. The only thing I can think about is that when you start the Pi in a remote area, it may not have the proper time and you may need an initial connection (router with cell phone) or set it manually with keyboard and monitor.
Ideally, I should add a setting in the portal to set the time manually in the GUI.

If your time is right but it's still not capturing images, try setting UPLOAD_IMG and UPLOAD_VIDEO to false in config.sh

These should be set to false by default but if you set them to true and you have no network, they will try to upload all the time and may make the system unstable.

drt102, 2018-12-14:
All UPLOAD_xxx options in config.sh are set to false so this is not the problem. It looks like it doesn't actually start capturing images so I think that you are right in that it cannot determine the time and so doesn't know whether to start capturing or not. Do you think adding a clock module (RTC) to the Pi would solve the problem?

sjwaame, 2018-12-14:
NTP service installation pointing to your router if it has a NTP server
option and your RPi is always on time

drt102, 2018-12-14:
My camera is battery operated and completely self-contained and I am looking for a solution that will allow me to capture images in a remote site with no additional connectivity.

thomasjacquin, 2018-12-14:
An RTC shield would do the job. Something like this is probably all you need: 
https://www.adafruit.com/product/3386

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issue appear to have been resolved or a resolution suggested.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys