AllskyTeam/allsky #48: 24 Hour capture

spacemanbobby, 2018-12-13:
What would I need to change to make the allsky program do 24hour captures and then convert them to the movie instead of just night time.

thomasjacquin, 2018-12-14:
You can try to the the angle value in the portal to 90. That means start/stop the capture when the sun is less than 90 degrees above the horizon (i.e. all the time).
This should take 24hr of images. The cut off between days happens at noon.

thomasjacquin, 2019-01-19:
I found that I need to add a bit of code to capture.cpp. Nothing major. I also want to do other changes to simplify the config.sh thing so I’ll tackle them both at the same time

ZebrasHaveBigEars, 2019-05-13:
I will try this setting and see how it works for me ... I have cloudy and city-lit nights so some daytime recordings would liven things up a bit! Cheers!

ZebrasHaveBigEars, 2019-05-14:
Yes I can confirm that the program does not run when angle value is set to 90.  I think my top two desired features are (a) Capture pictures during the day and (b) analyse for meteors. Having said this, the software is very very good already.

ZebrasHaveBigEars, 2019-05-14:
Mind you, set the angle to 89 and it seems to cheerfully run! I will see what the output is like over the next few days

spacemanbobby, 2019-05-14:
Please send us a link on a full day when it finishes please.

ZebrasHaveBigEars, 2019-05-14:
Well, it started but something went wrong and now something is wrong when I set the angle back to -6.  I hope I will have some over the next day or so to investigate.

spacemanbobby, 2019-05-15:
When ever I tried to get all day shots it would never record the day images
or combine them in time.  I think it is something in the logic and it also
takes a while to convert those images into startrails and MP4.  I would go
into the images folder and select delete so nothing is in there.. reboot
the device and watch the CPU %   mine would always Max out and then lock up
when I tried anything close to the 90-85 angle..  I eventually gave up on
an all day-night video.

thomasjacquin, 2019-05-16:
You're right, I need to add a bit of logic in the `capture.cpp` file. but feel free to experiment by adding this line in `saveImageDay.sh`

```
# Save image in images/current directory
cp $IMAGE_TO_USE "images/$CURRENT/$FILENAME-$(date +'%Y%m%d%H%M%S').$EXTENSION"
```

This is what you need in order to save the current image to disk. However, the timelapse still happens at dawn (or whatever angle you set the sun at). Ideally, for 24hr capture, we should start the timelapse at noon to have the night centred in the middle of the video.

ZebrasHaveBigEars, 2019-05-20:
Thanks for this code which nearly worked but it certainly showed the way to do it. I got it working by being inspired from the code in saveImageNight.sh:

1) Adding the commands to make the directory $CURRENT first where $CURRENT is the current day  as as opposed to the day before
2) Using %Y%m%d   as opposed to  %Y%m%d%H%M%S

and note

1) It is easy to check if it is all happening right by determining if the images go into the new folder under images
2) Making the movie of the whole day and the ftp of the whole day takes quite a lot of time as the new movie is quite large!

I am happy to share the code if needed.

spacemanbobby, 2019-05-21:
Please share the code if you wouldn't mind

ZebrasHaveBigEars, 2019-05-21:
yes sure .. please give me 24 hours as I took it outside for the first time last night ... I got the day and night video this morning! Thanks to Thomas!

bbillp, 2019-05-21:
Where do we post photos of our ALLSKY camera build ?  I designed and 3d printed mine but its still a work in progress while I figure out cooling and air circulation.    Bill in southern california.          PS: Thanks for the wonderful software.

ZebrasHaveBigEars, 2019-05-21:
I left photos of mine here:
https://www.instructables.com/id/Wireless-All-Sky-Camera/

ZebrasHaveBigEars, 2019-05-21:
Hi there

To save images during the day and night, here is the code which goes in saveImageDay.sh and which has worked over a 24 hour period for me.  It creates the directory under Images which will later be used in saveImageNight.sh and then copies the daylight images into it. Please remember to save a copy of the original version. Please be aware it takes a lot of time when the images are compiled into the movie and ftp is used to transfer the files to the website.  Keograms and startrails should also work.

Cheers

ZHBE

#########################################
## ADD THE CODE BELOW THIS LINE IN saveImageDay.sh
##cp $IMAGE_TO_USE "liveview-$FILENAME.$EXTENSION"
################## ADD THESE EXTRA LINES#############

CURRENT=$(date +'%Y%m%d')
mkdir -p images/$CURRENT
# Save image in images/current directory
cp $IMAGE_TO_USE "images/$CURRENT/$FILENAME-$(date +'%Y%m%d%H%M%S').$EXTENSION"
##########################################

ZebrasHaveBigEars, 2019-05-26:
Did you get it working? It is working a treat on my system.

spacemanbobby, 2019-05-26:
Haven't gotten to put the code in yet.. Am really busy with some other
things, but I will let everyone know how it comes out once I get the first
one online.

spacemanbob.com is my site.

CleverJay, 2020-02-07:
I added the code above to the saveImageDay.sh. Do I also need to set the angle camera setting to 90 degrees?

michaeloconnell78, 2020-06-26:
I tried this but it didn't work properly.
It saved individual images for the daytime but didn't create a video.
It also stopped recording at night time.

minnican, 2020-06-26:
This is brillianyt and just what I wanted. ZHBE I've just added the code to my saveImageDay.sh and restarted Allsky and it's saving images to the newly created /images/26062020 folder straight away. Is there anything else I need to do or will the time-lapse be created without further intervention please?

michaeloconnell78, 2020-06-28:
@ZebrasHaveBigEars I tried this and it saves the daytime images. Thanks!

@minnican It does not create a timelapse for daytime.

minnican, 2020-06-28:
@michaeloconnell78 If you leave the camera running 24-hours it creates the timelapse automatically when the sun is rising at -6 degrees below the horizon - in my case that's 03:45 hours at present. Otherwise you can create the timelapse manually by running the timelapse.sh in the scripts folder if I recall.

michaeloconnell78, 2020-06-28:
@minnican 
At present, it only creates a timelapse of the night videos.
I'll try running that script.
Thanks.

Skippyinspace, 2020-07-04:
Silly Question time.

Where do I add the code mentioned .

At the start, at the end, after a certain line?

Or is it a case of deleting the  Original file info and adding the code.

IF it cant be broken I will find a way to break it by accident.

minnican, 2020-07-05:
Here's the code in my saveImageDay.sh:
#!/bin/bash
source /home/pi/allsky/config.sh
source /home/pi/allsky/scripts/filename.sh

cd /home/pi/allsky

IMAGE_TO_USE="$FULL_FILENAME"
cp $IMAGE_TO_USE "liveview-$FILENAME.$EXTENSION"

# If upload is true, create a smaller version of the image and upload it
if [ "$UPLOAD_IMG" = true ] ; then
	echo -e "Resizing"
	echo -e "Resizing $FULL_FILENAME \n" >> log.txt

	# Create a thumbnail for live view
	# Here's what I use with my ASI224MC
	convert "$IMAGE_TO_USE" -resize 962x720 -gravity East -chop 2x0 "$FILENAME-resize.$EXTENSION";
	# Here's what I use with my ASI185MC (larger sensor so I crop the black around the image)
	#convert "$IMAGE_TO_USE" -resize 962x720 -gravity Center -crop 680x720+40+0 +repage "$FILENAME-resize.$EXTENSION";
##cp $IMAGE_TO_USE "liveview-$FILENAME.$EXTENSION"
################## ADD THESE EXTRA LINES#############

CURRENT=$(date +'%Y%m%d')
mkdir -p images/$CURRENT
cp $IMAGE_TO_USE "images/$CURRENT/$FILENAME-$(date +'%Y%m%d%H%M%S').$EXTENSION"
##########################################
	echo -e "Uploading\n"
	echo -e "Uploading $FILENAME-resize.$EXTENSION \n" >> log.txt
	lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION; bye" &
fi

I hope it helps a little. I should just say I have no idea why "# If upload is true, create a smaller version of the image and upload it" that is appearing as a much larger , bold font in my reply. It's just normal text.

Skippyinspace, 2020-07-05:
Thanks for the code, Ive noticed that on the pi the Image.jpg is updating and not the Live Image or Resize Image JPG which is what the Index page of the website uses and more importantly what the GUI uses as the live view.  I assume I just need to change a line from live-image.jpg to Image.jpg, but in which file!
Thanks

Skippyinspace, 2020-07-07:
Further to the my last message.
 I have changed the code to the 24 hour capture  / timelapse, but it only displays a live image on the GUI at the time of night time capture and doest seem to save the daytime images anywhere but the file image.jpg on the pi updates. Thanks Ian

minnican, 2020-07-07:
I'm afraid I don't use the GUI so it's difficult to say. I have 3 different images called "image.jpg", "image-resize.jpg" & "liveview-image.jpg" all being created in the "/allsky" folder. In addition to this I have images with names formatted like this "image-20200707035328.jpg" saved constantly in /allsky/images/20200707, the folder name representing the date of course. In order to see the live image without the GUI I have to click on the "liveview-image.jpg" or easier still just navigate to my website at http://www.minnican.com I admit that I have on more than one occasion, completely gone back to the beginning of the programming process to ensure that any tiny error is eradicated. I even re-flashed the SD card at one point. Worth it in the end I guess.

Skippyinspace, 2020-07-07:
> I'm afraid I don't use the GUI so it's difficult to say. I have 3 different images called "image.jpg", "image-resize.jpg" & "liveview-image.jpg" all being created in the "/allsky" folder. In [...]

- Thats whay=t should be happening, but the image-resize isnt  updating or the date.kpg isnt being created for day time, Its like the daytime capture switch has been turned ff

minnican, 2020-07-07:
> * Thats whay=t should be happening, but the image-resize isnt  updating or the date.kpg isnt being created for day time, Its like the daytime capture switch has been turned ff

At this point, because my programming knowledge is very poor, I'd go back to the beginning. I really didn't do anything different to what's in the this thread.

DarKOrange75, 2020-07-18:
Hi all, instead of a 24h video would it be possible to have one video for daylight and one for night?

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the original issue appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys