AllskyTeam/allsky #49: Image bit depth

dbainbridge, 2018-12-17:
I am trying to understand how we can take full advantage of the cameras available bit depth.  I am using a ASI178MC which has 14 bit ADC.  My morning pics show sharp gradients in the sky. I assume this is because it is only using 8 bits per channel when saved as JPG.

I have not looked into this yet but was wondering how can I test using full 14 bit/channel?

I was also slightly confused by the config settings:
`Image format. 0=RAW 8 bits, 1=RGB 24 bits, 2=RAW 16 bits`

It looks like the RAW is being described per channel but the RGB for all 3 channels?

thomasjacquin, 2018-12-18:
Hi David,

I believe the RGB output only works with 8bits per channel (3 x 8 = 24bits) and the greyscale output can give 8 bits or 16 bits images.
I think ZWO describes the image format this way to reflect the image size that gets written to disk. The RAW 8 bits is the smallest image, 16 bits takes twice the size and RGB takes 3 times the size. However since RGB is usually saved as JPG, we don't usually notice this difference.

To remove the gradients, you should try saving the image as a PNG. It will remove the compression and increase the file size but the result will be much better.

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the questions appear to have been answered.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys