AllskyTeam/allsky #52: Color image inverted

mcorretge, 2018-12-25:
Hello, 

I use an ASI120MC-S (USB 3.0) In RGB24 mode works well, also RAW8 image mode, but If I choose RAW16, changing the extension to .png and quality to 9, I don't see nothing, (I suppose that ASI120 isn't available to work in that resolution) so when I reset settings to default values with RGB24 image type, now the color of image was inverted (in RBG24 https://www.dropbox.com/s/ijciuupro1uf77s/1.jpg?dl=0) and (in RGB24 after try RAW16 https://www.dropbox.com/s/uofanygpqa3y1cc/2.jpg?dl=0)I could reinstall all, but I will lose the configuration, if there is any fastest solution, I've tried to restart

Thank you, 

Marc

thomasjacquin, 2018-12-26:
Hi Marc,

If restarting the service and rebooting the Pi didn't fix it, I have a feeling it's got something to do with the camera itself. It seems like a buffer issue. Have you tried going back to RAW8 and then RGB24?

mcorretge, 2018-12-28:
Thanks Thomas,

I found the problem, and was the dark frame subtraction option.

I was testing the dark frame option and I didn't read well the instructions and activate the dark option for a long time, and when I deactivate the option the dark image "dark.jpg" these file was the same as a normal image, and for that I saw the inverted colors.

The problem was that I leave the option activated for a long time instead of activating only for a frame and then deactivate.

If i remove dark.jpg file the problem is solved

Thanks

thomasjacquin, 2018-12-29:
Great. Yes, I should add a warning somewhere to notify users that dark frame subtraction is on.

closed 2018-12-29 by thomasjacquin