AllskyTeam/allsky #53: RPi zero W doesn't work

SRadiant, 2018-12-28:
Hi, everyone. I have a problem when I run AllSky in RPi zero W. It doesn't work. I sure that the OTG cable is fine (keyborad and mouse OK) and ASI120 is power on. But it doesn't take any image. Does anyone know how to make it working? Help me, please. Thanks!

thomasjacquin, 2018-12-29:
What's the output in the terminal when you run the software manually?
`./allsky.sh`

SRadiant, 2018-12-29:
I had solved the problem. 

In README.md...
Now, before running the install script, if you're running a Pi 2, it may not be compatible with armv7 architecture. Run cat /proc/cpuinfo to know your processor model. If it doesn't say ARMv7, you'll need to change the first line of Makefile to say platform = armv6

Thank you, Thomas.