AllskyTeam/allsky #56: ftp path

spacemanbobby, 2019-01-19:
I have my image-resize uploading to my directory but for some reason at the end of the night the mp4 and the startrails and keogram all upload to that same directory. 

how would the pathing look for it to go to those directories. I obviously have enabled the uploads since they are all going to my site but they are not making it to the directories they need to be in.. I have to manually ftp and move them each day.

thomasjacquin, 2019-01-19:
Did you set the correct path in scripts/ftp-settings.sh ?

spacemanbobby, 2019-02-16:
> I'm running an USB pen drive 256Gig in my Wi-Fi access point with
> following ftp settings. You need to validatie Paths are correctly set.
> Note with me it is all Case Sensitive.
> I created the directory AllskyFTP and the corresponding sub directories on
> my pen drive.
>
> !/bin/bash
> FTP/SFTP settings
>
> PROTOCOL='ftp'
> USER='xxxxxx'
> PASSWORD='xxxxxx'
> HOST='192.168.1.xxx'
> IMGDIR='/KINGSTON/AllskyFTP/Images'
> MP4DIR='/KINGSTON/AllskyFTP/Videos'
> KEOGRAM_DIR='/KINGSTON/AllskyFTP/Keograms'
> STARTRAILS_DIR='/KINGSTON/AllskyFTP/Startrails'

Did you have to create those directories before you started the system up
or where they created when the program needed to access them.

thomasjacquin, 2019-02-17:
You would need to create these directories manually on your host otherwise the upload will fail and lftp will return an error message.

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved or at least a resolution suggested.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys