AllskyTeam/allsky #57: Timelapse avconv issue

Seb-drocks, 2019-01-20:
I checked the previous issue of timelapse not generating and my problem is different. I get this error when i run the script manualy:
avconv: symbol lookup error: /usr/lib/arm-linux-gnueabihf/libflite_cmu_us_awb.so.1: undefined symbol: cmu_us_awb_f0_ah_42_cart
I use a RPI 3b and there are no errors until it use the avconv command.

thomasjacquin, 2019-01-20:
Hi Seb,

I'm not sure why it's giving you that specific error. Looks like `libflite` is a text-to-speech library. It must be bundled with ffmpeg.
Maybe you can try uninstalling `libav-tools` and re-installing it.
`sudo apt-get purge libav-tools`
`sudo apt-get install libav-tools`

Hope that fixes the issue. Do you have any other software installed on the Pi that could conflict with the timelapse generation?

Thomas

Seb-drocks, 2019-01-20:
Sadly it didn't even let me unistall.
Segmentation fault
E: Le sous-processus /usr/bin/apt-listchanges --apt || test $? -lt 10 a renvoyé un code d'erreur (1)
E: Failure running script /usr/bin/apt-listchanges --apt || test $? -lt 10
I tried installing still and reconpile and still the same error.
The only thing I could think of right now is because i installed raspbian in french ? From what I know it shouldn't affect the commands and i only installed the allsky related stuff from the github.

thomasjacquin, 2019-01-20:
The french version of raspbian shouldn't be an issue.
However, the fact that you're getting a segmentation fault with `apt` seems to indicate that the problem is not related to the allsky software but it's a more general issue at the OS level.
If that's not too much work for you, I would re-install raspbian and see if you're still getting the error with the allsky software.

Seb-drocks, 2019-01-20:
I will try that.

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys