AllskyTeam/allsky #58: GUI live view showing black Image

spacetrace1982, 2019-01-20:
Hello. 

I seem to have intermittent issues with the GUI live view showing black images. I have had it working perfectly with the text at the top corner as expected with an image from the camera. Then I often get the instance whereby I only get black images displayed with no text overlay, so I know that it is not just a too short of an exposure at night. I have tried:

restart rasp pi
restart gui page
restart the all sky manually using terminal 
checked the status of the all sky cam using terminal
restart the internet just in case

to double check on this I go to the raspberry pi and check preview of which I see that it is capturing an image as the preview shows me an image from the camera. 
Attached screenshot of settings from GUI 
Also included a screen shot of the live view screen 

[image]

[image]

thomasjacquin, 2019-01-21:
Hi, 

The only thing in the code that would generate completely black frames is if you’re taking a dark image. In this case, the overlay is not applied to the image.

Also, when using autogain and auto exposure at the same time it can result in black images.

Do tou get black images in a row or is it only once in a while?

spacetrace1982, 2019-01-21:
It is black images in a row. I am certain that is is due to location settings. I reverted back to the default location and it worked OK, I put in my location and I got a black image, so I assume that it thought that my location was still sunny side up even though it was night. Is there a way to view the image in daytime too? This morning, same again, black GUI live screen, nothing from the camera is being displayed. 

Thank you

The-J-BB, 2020-05-31:
Hi there!

Since yesterday, I have the same issue with black images. If I don't stop the service, the images will be exposed until the longest setted time (daylight). Actually Re start did't worked, also dark enable and disable, also reboot...

jcauthen78, 2021-02-04:
I'm having the exact same problems on my PI-4b. During the day, the local web gui live view comes up black just like shown above. I watch the file sizes in the /allsky folder, and they change from time to time by very small amounts, but i think it's just random compression artifacts because the "image.jpg" file stays the exact same, almost like its stuck. 

Reboots haven't fixed it, start/stop service hasn't either. As above, if i change my LAT/LON to where it's dark, it runs as normal without issue... it seems to be ONLY when its daytime in the LAT/LON coordinates.  I made sure the Auto Exposure and Auto Gain aren't on at the same time as well. 

This isn't a major issue other than those of us trying to setup our systems during the daytime after initial builds (camera/box rotation for correct imaging). Sun is just setting here, so I should get the local web GUI back running as normal and won't be able to troubleshoot until tomorrow. 

Any suggestions? 

P.S. the code editor on the GUI.... I love you!  makes life so much easier!

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software.  If you think it might still be relevant in version 0.8, please submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys