AllskyTeam/allsky #59: Best camera settings for ASI120MC?

triplanedave, 2019-01-23:
Thomas,

The following MP4 is pretty representative of most of clear sky images.  https://1drv.ms/v/s!Aqml0ppAJEnxl81lEdhvhoLAQIm_dQ

I feel like there are some settings that could improve my overall quality, but not sure if it's obvious to you what my best changes would be.

Here's my settings.json text:

{"width":"0","height":"0","exposure":"20000","maxexposure":"20000","autoexposure":"0","gain":"70","maxgain":"200","autogain":"1","gamma":"70","brightness":"50","wbr":"53","wbb":"90","bin":"1","delay":"10","daytimeDelay":"5000","type":"1","quality":"95","usb":"40","filename":"tcosky.jpg","flip":"0","text":"text","textx":"15","texty":"30","fontname":"0","fontcolor":"255 255 255","fonttype":"0","fontsize":"0.7","fontline":"1","latitude":"47.1N","longitude":"120.77W","angle":"-6","time":"1","darkframe":"0","showDetails":"1"}

thomasjacquin, 2019-01-23:
Hi David,

It doesn't look like auto-gain was set for this video or maybe it didn't work expectedly.
Things I would change are: 
- increase the max-exposure to maybe 30s, 
- enable auto-exposure, 
- disable auto-gain, 
- set the gain to 200 or higher
- lower gamma to 50,
- Take a dark frame (while covering the dome) and apply dark frame subtraction

Start with that and we can tweak things later for your sky conditions.
What lens are you using?

Thomas

triplanedave, 2019-01-23:
Thanks Thomas.  I'll set those values and try tonight.  I am 1 1/2 hours drive from the camera, so I can't do the dark frame tonight, but will try this weekend when I'm onsite.  I'm using the CCDTV lens that comes with the 120MC.

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys