AllskyTeam/allsky #60: Basics in taking pictures and doing timelapse

Vigo79, 2019-01-24:
Hello, Since my fat problem the last days I´m ready to taking pictures. I think it´s a silly question, but I don´t found an answere...
1. When does the allsky software start taking images? I see no "start" / "Stop" button. 
2. When I look in the GUI => Images => there are maybe 20 picture from before 15 minutes. But no more. Why does it stop taking pictures? 
3. In the GUI => Images  I have timelapse, Keogram and Startrails. But they are empty. 
In the Code/Manual Page here in Github I see at the end how to create them manually with terminal. Is there a choice to do it with the GUI? 

Sorry for my bad english... :D

Bye Vigo

thomasjacquin, 2019-01-24:
The software runs all day but it only saves images during the night. More specifically between dusk and dawn.
This can be adjusted with the `angle` setting in the GUI. It is the angle of the sun below or above the horizon at which the capture should start and stop.
During the day, images are captured as well in order to display them in the live view but they are not stored anywhere.
Make sure your RPi has the correct time and timezone.

There's no options on the GUI to generate timelapse, keograms and starttrails but I may add it in the future.

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys