AllskyTeam/allsky #61: Editing the allsky portal GUI webpages - Night time friendly colors

rave3c0, 2019-01-26:


thomasjacquin, 2019-01-28:
Hi Bill, most of the CSS changes can be applied in 2 files:
- dist/css/sb-admin-2.css
- dist/css/custom.css

I would try to put most of the overrides in the second file to keep all changes at the same place.
I just did a quick test in my browser and I think a dark theme would look good. It wouldn't take much time for me to write the changes. I'll try to get it done this weekend.

[image]

rave3c0, 2019-01-30:
Thanks Thomas!  I dig the look already, much appreciated.

rave3c0, 2019-02-04:
Awesome!  I have had a few days and nights with the new GUI and its been great plus easier on the eyes!  Thanks for adding this feature.  Now i can sit outside and check on the pics or focus and still retain some night vision to admire the view.   I am using my only astro cam for this project so I do a lot of refocusing when switching between telelscope and allsky... getting the most of what i got ;) Thanks for the contribution! 

One minor issue, the theme switcher only works on firefox, a mess on Chrome.  On Chrome the light dark mode isn't clickable and some other formatting is off.  Just FYI. Everything works in Firefox.

[image]

[image]

browser versions: 
Chrome 71.0.3578.98 x64
Firefox Quantum 65.0  x64

thomasjacquin, 2019-02-04:
Hey Bill,

Thanks for the feedback. I am using Chrome at home and it seems to work fine. Can you try clearing your browser's cache? Ctrl+Shit+delete is the shortcut I think.
I'm guessing some of the old CSS is still cached in Chrome.

rave3c0, 2019-02-07:
That did it, cleared out the cache. Works great, even on mobile too!  

Thanks again for the update.  Love this project, Ive been getting some great images and star trails.

closed 2019-02-08 by rave3c0