AllskyTeam/allsky #62: Image breaks down during the night

mcorretge, 2019-01-27:
Hello Thomas,

I have the new raspberry pi B+ with ASI120MC-S, during the day works well, but in the middle of the night the image it starts to be wrong, during the day and first time of night https://www.dropbox.com/s/9twaqvisdb57vud/0595.jpg?dl=0 and suddenly during the night the image breaks down like this https://www.dropbox.com/s/2n0b6wsmtilnqbh/0835.jpg?dl=0

https://www.marccorretge.com/allsky/videos/allsky-20190126.mp4 this is the timelapse of the last night where you can see how the image suddenly breaks down

Sometimes solves itself, or at the beginning of the day also solves itself.
How can I fix it?

Thanks.

thomasjacquin, 2019-01-28:
Hey, I've never seen this kind of issue before. From the timelapse, it's hard to tell if it's creating a new image or if it's looping over the same one for the rest of the night.
Do you notice it more when you use a special camera setting?

gremlinforce007, 2019-04-14:
Hi Thomas, 

I've noticed tonight i'm having the same issue as well.. have to reboot the camera to fix it ..

thomasjacquin, 2019-04-15:
@gremlinforce007 Can you post your settings. I can take a look to see if I see anything obvious. It could come from a number of reasons (camera model, SD card, usb cable, camera settings, etc).

gremlinforce007, 2019-04-15:
@thomasjacquin : here you go .. thanks for your help 

[image]
[image]

thomasjacquin, 2019-04-15:
When I use auto-exposure and auto-gain together I found that it produces unpredictable behaviour. I think it's something coming from the ZWO library.
Also I haven't really tested binning 2 so I will take a look on my camera to see it I can reproduce the problem.

gremlinforce007, 2019-04-18:
So update:

Whenever Auto Gain goes above 64 i see this issue occurring :(

gremlinforce007, 2019-04-24:
Hi @thomasjacquin 

Its happening again really bad last night, 
Any ideas ? 
https://skypulse.org/allsky/keograms/keogram-20190423.jpg
https://skypulse.org/allsky/videos/allsky-20190423.mp4

[image]
[image]

mcorretge, 2019-04-24:
It keeps happening to me from time to time, there are nights that work perfectly, but there are other nights like this one [https://marccorretge.com/allsky/keograms/keogram-20190419.jpg](url)
As someone said the solution is to restart the service, but at midnight we will not be waiting to restart it every time. I read in a blog that the last ASI120 had some error in the firmware that gave several problems, in fact the manufacturer of the ASI (ZWO) have stopped manufacture it and it could be that it had something to do with these errors that it gives us, it would be interesting to know if it happens with any other model than ASI120.
The good thing is that after a while it is correct again, anyway, I in the EndOfNight file in the last line I put (sudo service allsky restart) to make sure that if it is not corrected automatically that does it automatically every morning. Also is interesting.
Also it only happens to me when it is night, during the day it does not give this problem

mcorretge, 2019-04-24:
https://marccorretge.com/allsky/keograms/keogram-20190419.jpg

thomasjacquin, 2019-04-26:
Unfortunately , I don't own an ASI120 camera so I can't replicate the issue at home. It seems to me that this is a firmware or hardware issue with this model. Have you tried the firmware upgrade tool that they provide on their website? I am wondering if the problem happens with both versions of the firmware.

Chboss74, 2019-06-10:
I had something similar with my ASI120, so I changed changed the USB cable and reduced the cable length, which improved things. However I had some issues with pictures not being debarred occasionally and then permanently. Did a complete reinstall of the Raps including allsky software and seems to work properly now.

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys