AllskyTeam/allsky #63: Startrails issue

jasonvrhodes, 2019-01-27:
Hello, everything is working great, the only problem is the making of the startrails.. 
in the morning in the page of the startrail i find only one image of the sky.. but no star trails.. i tried to do it manuali writing the command, but it does the same... where are I getting wrong? 
Thanks for any help
Jason

thomasjacquin, 2019-01-28:
Hi Jason,

The startrails feature works by stacking all images below a certain brightness threshold.
The default is 15% overall brightness.
This works really well when using a fixed exposure and fixed gain, however, if auto exposure is used, the image will always have a moderate brightness (probably higher than the threshold).

The reason for this threshold is to discard overly bright images (cars, moon, etc) that would ruin the startrails.

You can still try to generate the startrails manually by using different values for the threshold (0.5, 0.7, etc) but it will probably end up a bit over-exposed.

Hope that makes sense.

jasonvrhodes, 2019-01-29:
Thanks, Thomas, i will try this night...  my settings before was with fixed 10 sec. exposure  and gain fixed ad 50, i will try using less seconds of exposure and a lower gain.
Thanks 
Jason

thomasjacquin, 2019-01-29:
Hey Jason, 10 sec at 50 gain should work fine. Maybe you just need to increase the threshold.

You can do that by increasing the BRIGHTNESS_THRESHOLD value in `config.sh`. Try 0.2 or 0.3

jasonvrhodes, 2019-01-29:
Thomas!! GREAT!!! it worked! 
thank you for this great program!! you made my day Perfect!!!!

Timburgess1, 2019-09-18:
What’s the command I need to run to create a startrail after modifying the threshold?

thomasjacquin, 2019-09-18:
Change the parameters in the following command to match your paths and threshold.
Run from /home/pi/allsky
Params are 

1. Location of your images
2. Extension
3. Threshold
4. Output Destination

`./startrails ./images/20180223/ jpg 0.15 ./images/20180223/startrails.jpg`

Timburgess1, 2019-09-18:
Fantastic, thanks for your help.. it’s a super little project.. on night 4 tonight..

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys

neurosurgeon, 2023-09-16:
> Hi Jason,
>
> The startrails feature works by stacking all images below a certain brightness threshold. The default is 15% overall brightness. This works really well when using a fixed exposure and [...]

But how do you actually re-generate the star trails? What's the command and where do you enter it?

EricClaeys, 2023-09-16:
@neurosurgeon 
Run:
~~~
cd ~/allsky/scripts; ./generateForDay.sh --startrails DATE
~~~
replacing "DATE" with one of the directories in ~/allsky/images, e.g, `20230915`.
Run:
~~~
cd ~/allsky/scripts; ./generateForDay.sh --help
~~~
to see the other options.
When I test startails here's what I do:

1. `cd ~/allsky/images`
2. `mkdir test`
3. Copy about 100 nighttime images from some directory, e.g., 20230915 to `test`.  This makes testing MUCH faster.
4. Edit "config.sh" via the Editor page in the WebUI.
5. Look for `BRIGHTNESS_THRESHOLD`.  Keep it open in one window.
6. `cd ~/allsky/images/test`
7. `../../scripts/generateForDay.sh --startrails test`
8. Rename `startrails.jpg` to `startrails-BRIGHTNESS_THRESHOLD.jpg` replacing "BRIGHTNESS_THRESHOLD" with the value in the file.
9. Update BRIGHTNESS_THRESHOLD
10. Repeat steps 7, 8, and 9 until I find something I like.
11. `../../scripts/generateForDay.sh --startrails DATE`  to remake last night's startrails. 
~~~

neurosurgeon, 2023-09-18:
Thank you! It worked and I can finally see star trails. No if I can only
get my HOA to turn off the annoying street lamps.

Regards,
Ausaf

EricClaeys, 2023-09-18:
@neurosurgeon    Great!!  Glad it's working.
What are you using for BRIGHTNESS_THRESHOLD ?

neurosurgeon, 2023-09-18:
It’s working at 0.2-- Ausaf A. Bari MD PhD FAANS FACSAssociate ProfessorDirector of Functional and Restorative NeurosurgeryDavid Geffen School of MedicineSurgical Neuromodulation and Brain Mapping LabUCLA NeurosurgeryEmail: ***@***.***

Stephen2615, 2024-05-02:
@EricClaeys 
Hi Eric, sorry to write something on an old thread but I have been experimenting with star trails. The default (0.1)never produced any star trails. After a lot of experimentation with and without a full moon, I found 0.25 to be a decent setting. Moonless nights without cloud produces long star trails. With a bright moon and clouds, the star trails are still visible amongst the other rubbish.  I got to wondering if there was some way of using that stats to make the threshold dynamic but in all reality, it didn't seem worth the effort. If anything, a mid point between minimum and mean seemed to be something that on occasion produced better full moon nights. 
Cheers
Stephen

EricClaeys, 2024-05-02:
If you had a way to determine the % moon showing at your location it would probably be easy to convert that to a threshold, and very easy to update the threshold setting.  That script could be run every day so the startrails command always had an up-to-date number.

If someone gave us the % moon code we could fairly easily add the Allsky changes.  I'd probably change the single startrails threshold into "new moon" and "full moon" thresholds, and the actual threshold used would used the % moon to interpolate a threshold between those two numbers.

Stephen2615, 2024-05-02:
@EricClaeys 
Interestingly, one of the Allsky variables is ${MOON_ILLUMINATION}. I don't know how to call it as it doesn't work in the overlay editor and it doesn't seem to be exported into a json file.

EricClaeys, 2024-05-19:
@Alex-developer, when we make startrails generation a module rather than having it be part of endOfNight.sh, will it have access to the MOON_ILLUMINATION variable, so it could use that value in adjusting the brightness threshold?