AllskyTeam/allsky #65: no ftp upload on bplaced.net

nemichael, 2019-02-04:
Hi,

i cant get it to wrok, to upload a image or startrail or the others to my webspace.

here is my ftp_settings.sh

#!/bin/bash

# FTP/SFTP settings
PROTOCOL='ftp'
USER='*****'
PASSWORD='*******'
HOST='wetter-wien.bplaced.net'
IMGDIR='/www/wetter/allsky/'
MP4DIR='/www/wetter/allsky/videos/'
KEOGRAM_DIR='/www/wetter/allsky/keograms/'

here is the log file
....

1268170
Saving image-20190203234645.jpg

Resizing image.jpg

Uploading image-resize.jpg

1882347
Saving image-20190203234704.jpg

Resizing image.jpg
......

he talt me that he has uploaded it but on my server there is no image. did i need a special ftp programm or shoud it work out of the box?

on the webgui i see all images but the images will not be uploadet?

regards
michael

thomasjacquin, 2019-02-06:
Hi Michael,

`IMGDIR`, `MP4DIR` and `KEOGRAM_DIR` are paths on your server.
It could be something like `/lamp0/web/vhosts/www.wetter.com/htdocs/allsky`
You can find the exact location using filezilla or cpanel.

Thomas

flyndr, 2019-02-06:
I have similar problem. i gather we have to make these directories on the host computer?

thomasjacquin, 2019-02-06:
Yes, these directories need to be created on the host otherwise the ftp upload will fail. Maybe I should mention this somewhere in the README

flyndr, 2019-02-06:
Further to above. we have solved FTP issue by enabling ssl. Files are uploading to sever folders now.
Next question is how do we display on website. Can do the pictures ok but i would like to add the control buttons you have on your site. Can you help us with a way to do this?
Also, our observatory is in the country with limited broadband (we pay for every mb). I have not had a good look yet but is there a way so set upload to 10 or 15 minutes.
Thank you for an elegant piece of software

Terry

thomasjacquin, 2019-02-06:
You can use the code I wrote for my website. You'll only have a few values to change in `controller.js` as explained in the README file.
https://github.com/thomasjacquin/allsky-website

As for the software, it currently uploads everytime it captures a new image but it's a feature that other users have mentioned before. I'll see how much work needs to be done for this feature.

flyndr, 2019-02-06:
Thank you Thomas. I think we have enough to complete the job now. Night here so will look at it in yjhe morning. Will let you know one way or the other how it is going.
Thanks again for your time and help.

Terry

flyndr, 2019-02-06:
Thomas Thank you. That last link is what we were looking for. We found it hard to find. Maybe a link on the main page under FTP settings? It is up and running now but still experimental as not yet built because I am waiting for right angle usb lead to fit in the dome. I will send you the link when up and running. Adjustable upload time may be a useful feature for  future development.
This device is for Canterbury Astronomical Society to check weather at the West Melton Observatory. This is a not for profit society. I would like to make a personal contribution to this project as you have been very patient and helpful. How do I do this?
Kind regards
Terry

flyndr, 2019-02-06:
A further note,
It looks like by setting daytime and night delay you can reduce the upload frequency.
Terry

thomasjacquin, 2019-02-06:
Hi Terry,
I am planning on having an auto-deploy script to install the website directly on the host (creating files, config and directories automatically). This could be called either in command line or from the GUI. That would be more convenient for everyone.
I'm happy to see the project has spread all the way to New Zealand!
For the time being, the daytime and nighttime delays should work for you but it adds delay between frames and you may miss some action (meteors, satellites, aurora, etc).

Here's how it behaves with delays (i=image, u=upload):
```
i----------i---------i----------i----------i
u----------u---------u----------u----------u
```

Ideally, it should capture without delays but only upload every X seconds or images:
```
i-i-i-i-i-i-i-i-i-i-i-i-i-i-i-i-i-i-i-i-i-i-i
u----------u---------u----------u----------u
```
In the latter scenario, you capture the whole night (nice to have if you do startrails, keograms and timelapse) but you save on your bandwidth.

There should be a donation button at the top of the instructions if that's what you had in mind https://github.com/thomasjacquin/allsky

Thanks!

flyndr, 2019-02-06:
Thanks again for your time

Terry

 

This email has been checked for viruses by AVG.
https://www.avg.com

nemichael, 2019-02-07:
hi,

now i get this 

> cd: Fataler Fehler: Certificate verification: Not trusted (22:ED:AA:C0:9F:0B:97:65:A2:20:B2:73:EB:52:5E:26:75:EE:B3:12)
Feb 07 21:35:45 allsky allsky.sh[2450]: put: image-resize.jpg: Fataler Fehler: Certificate verification: Not trusted (22:ED:AA:C0:9F:0B:97:65:A2:20:B2:73:EB:52:5E:26:75:EE:B3:12)
Feb 07 21:36:04 allsky allsky.sh[2450]: cd: Fataler Fehler: Certificate verification: Not trusted (22:ED:AA:C0:9F:0B:97:65:A2:20:B2:73:EB:52:5E:26:75:EE:B3:12)
Feb 07 21:36:04 allsky allsky.sh[2450]: put: image-resize.jpg: Fataler Fehler: Certificate verification: Not trusted (22:ED:AA:C0:9F:0B:97:65:A2:20:B2:73:EB:52:5E:26:75:EE:B3:12)
lines 1-14/14 (END)

> 

is there a way to configure the ftp to trust it?

regards
michael

thomasjacquin, 2019-02-07:
Hey Michael, I think there's a way to auto-accept certificates with lftp but I usually do this instead:

`ssh place.bplaced.net`

This should ask you if you want to accept the certificate and then you shouldn't have any issues.

nemichael, 2019-02-07:
hi,
thanks for your help put what should i do with ssh place.bplaced.net? write it in the config or what?

sorry for my bad english my mother language is german.

regards
michael

thomasjacquin, 2019-02-07:
Sorry, I just realized you're using ftp and not sftp.

In that case, you should probably be fine. All you need to do is replace
`	lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION; bye" &
`
by
`	lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set ssl:verify-certificate false; set net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION; bye" &
`
You will need to replace if in scripts/saveImageDay.sh, scripts/saveImageNight.sh and scripts/timelapse.sh

nemichael, 2019-02-07:
hi,
no sorry dont work

> cd: Fataler Fehler: Certificate verification: Not trusted (22:ED:AA:C0:9F:0B:97:65:A2:20:B2:73:EB:52:5E:26:75:EE:B3:12)
Feb 07 22:33:28 allsky allsky.sh[8800]: put: image-resize.jpg: Fataler Fehler: max-retries überschritten (EPSV failed, will try PASV)
Feb 07 22:33:37 allsky allsky.sh[8800]: cd: Fataler Fehler: Certificate verification: Not trusted (22:ED:AA:C0:9F:0B:97:65:A2:20:B2:73:EB:52:5E:26:75:EE:B3:12)
Feb 07 22:33:39 allsky allsky.sh[8800]: put: image-resize.jpg: Fataler Fehler: max-retries überschritten (EPSV failed, will try PASV)

> 

saveImageDay.sh looks now this

> #!/bin/bash
source /home/pi/allsky/config.sh
source /home/pi/allsky/scripts/filename.sh

cd /home/pi/allsky

IMAGE_TO_USE="$FULL_FILENAME"
cp $IMAGE_TO_USE "liveview-$FILENAME.$EXTENSION"

# If upload is true, create a smaller version of the image and upload it
if [ "$UPLOAD_IMG" = true ] ; then
        echo -e "Resizing"
        echo -e "Resizing $FULL_FILENAME \n" >> log.txt

        # Create a thumbnail for live view
        # Here's what I use with my ASI224MC
        convert "$IMAGE_TO_USE" -resize 962x720 -gravity East -chop 2x0 "$FILENAME-resize.$EXTENSION";
        # Here's what I use with my ASI185MC (larger sensor so I crop the black around the image)
        #convert "$IMAGE_TO_USE" -resize 962x720 -gravity Center -crop 680x720+40+0 +repage "$FILENAME-resize.$EXTENSION";

        echo -e "Uploading\n"
        echo -e "Uploading $FILENAME-resize.$EXTENSION \n" >> log.txt
        lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set ssl:verify-certificate false; set net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION; bye" &
fi

> 

regards
Michael

thomasjacquin, 2019-02-07:
What is your ftp server's address? I'll try to ping it from here.
I have tried `wetter-wien.bplaced.net` but it doesn't seem accessible.

nemichael, 2019-02-07:
hi

try this

http://wetter-wien.bplaced.net/wetter/

thomasjacquin, 2019-02-07:
That's the location of your website but you server should be configured to have an FTP account with user, password and host. It is sometimes like `ftp.nameofmyserver.com`

nemichael, 2019-02-07:
this is the adress i use in fileziller: wetter-wien.bplaced.net

flyndr, 2019-03-14:
Hi Thomas,

This is just for your information. It may help you to help others.

I have the allsky camera working now and visible on the website. I am very happy with its performance. A very nice device.

One problem I had was that it was working fine at night but when switching to daytime mode it just locked up.

I soon discovered that the necessary long power lead was too small and there was sufficient voltage drop to stall the Pi.

The ASI camera draws quite a bit of current (presumably more in daytime mode).

If others have this problem it is worth considering a lead with larger conductors.

Thanks again for all your help in the initial setup

Kind Regards

 

Terry Richardson

  <https://github.com/notifications/beacon/AtGV0Gxk9IhrApfSOKTycQL3EUzm8KR5ks5vKo6mgaJpZM4aga3K.gif> 

---
This email has been checked for viruses by AVG.
https://www.avg.com

thomasjacquin, 2019-03-15:
Thanks Terry, that's great information. I never had this problem myself but it's good to know it can cause issues.

bbillp, 2019-03-15:
Micro USB cables are a problem, use an ohm meter to measure the round trip cable resistance or a volt meter to measure the voltage drop across the cable.  Using an old USB cable cut the end off and wire the red/black directly to the Pi circuit board.  You will find the red/black power wire in the old cables to be robust, about 18 gauge.  Its really the easiest and most affective way to circumvent the tiny micro connector.  Another option is to adjust the power supply for 5.4 volts to overcome cable resistance.
Bill acj

flyndr, 2019-03-27:
Hi Thomas,

I am Observatory Director for Canterbury Astronomical Society.

I have built the allsky camera and it is now installed at our observatory as follows:-

R F Joyce Obsrevatory

West Melton

Christchurch

New Zealand.

GPS 143.3S, 172.3E

Please feel free to add this to your map (although New Zealand does not seem to feature)

This can be viewed at richardson.geek.nz/allsky

I have one question. Our members are ecstatic about the camera as the observatory is half an hour out of the city and a long drive if there are no stars. All are very enthusiastic about the camera but now I am getting gentle complaints that the overlay is not correct. My son has set up the server and has not quite got the coordinates right yet (he is in Auckland). Does setting the co-ordinates correctly on the website (the camera icon) set the overlay for local sky, or do we need to change something in the server?

I would be grateful for your help with this final hurdle.

Thanks again for a great piece of software

Regards

Terry

---
This email has been checked for viruses by AVG.
https://www.avg.com

thomasjacquin, 2019-03-27:
Hi Terry,

The only thing you need to modify is the latitude and longitude. The config is expecting numeric values.
```
...
latitude: -43.3,
longitude: 172.3,
...
```
The webpage will pick up the sign and place N, S, E, W accordingly.

EricClaeys, 2021-09-04:
Thomas, has any progress been made with the feature to allow images to be uploaded every X times, or every X minutes?  You discussed this about 2/3 the way up this stream.

EricClaeys, 2021-10-06:
Closing issue.  Request to upload images every X times was added to Projects list.

closed 2021-10-06 by EricClaeys