AllskyTeam/allsky #66: Software doesn't stop taking images

stauders, 2019-02-25:
The software didn't stop taking pictures even after the sun had risen for a long time. The angle was set to -6 degrees and the coordinates of the location were adjusted. Also the other images like Keogram and Starstripes were not generated.

thomasjacquin, 2019-02-25:
Can you send show me your settings? either screenshot of the GUI or copy of /var/www/html/settings.json

stauders, 2019-02-26:
`{"width":"0","height":"0","exposure":"1500","maxexposure":"10500","autoexposure":"1","gain":"60","maxgain":"151","autogain":"1","gamma":"50","brightness":"50","wbr":"53","wbb":"90","bin":"1","delay":"3000","daytimeDelay":"49999","type":"1","quality":"95","usb":"52","filename":"image.jpg","flip":"0","text":"","textx":"15","texty":"35","fontname":"0","fontcolor":"255 255 255","fonttype":"0","fontsize":"1","fontline":"1","latitude":" 49.99N","longitude":"8.24W","angle":"-3","time":"1","darkframe":"0","showDetails":"1"}`

thomasjacquin, 2019-02-26:
Try setting autogain to 0 or autoexposure to 0. They don't work together very well. This is some kind of known issue. I'm not sure that's what causing the problem but that would help debugging the issue.

stauders, 2019-02-28:
Sry ,had wrong location entered (west instead of east)

closed 2019-02-28 by stauders