AllskyTeam/allsky #67: video creating not execute everytime

dirkvahle, 2019-03-09:
Many thanks for your AllSky project. Im happy! :1st_place_medal: 

My web side [www.eberdingen.space](url) is running like yours. :-) But after some nights has my Rasbian Pi a problem with creating of video.

In the graphical interface and in settings.json i have write the filename allsky.jpg. If the project record a picture, then is it on my platform with a numerical file name. The list of files start from **0001.jpg** and end with **xxxx.jpg**. Here I you can see a pic from list: [www.eberdingen.space/allerlei/list_pictures_with_numeric.png](url) . The video on next day is fine.

But sometimes are the pics renamed in file names with **allsky-datetime.jpg**. You can see here this list: [www.eberdingen.space/allerlei/list_pictures_with_name+date+time.png](url) . In this case the video on next day isnt create.

I have try to find the mistace in the scripts, but i cant find it. Have you a solution of my problem?

thomasjacquin, 2019-03-09:
Hi,

The pictures are stored on the Pi with this format: allsky-datetime.jpg. At the end of the night, the endOfNight.Sh script is run and it renames the files in a sequence 0001.jpg, 0002.jpg. This renaming is needed for avconv to generate the timelapse. If tge files haven’t been renamed, this means there was an error earlier. Didi the startrails and keogram generate properly?

dirkvahle, 2019-03-09:
Startrail and kreograms files and folders was also not create.
With which command will the data be renamed?

Thanks

thomasjacquin, 2019-03-11:
The files are renamed line 21 of timelapse.sh
https://github.com/thomasjacquin/allsky/blob/master/scripts/timelapse.sh

But if the keograms and startrails were not generated, the problem is probably happening earlier.

dirkvahle, 2019-04-22:
I think my problem is solved. After changing of my website provider the problem did'nt occur anymore.

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys