AllskyTeam/allsky #69: ZWO 174MM 'max res' isn't max

maphilli14, 2019-03-12:
The camera should be 1920x1080 but the images are saved as 960x540 :( how come?
[image]

maphilli14, 2019-03-12:
In fact, the resolution changed when i turned gain off: https://www.youtube.com/watch?v=XvTiIcNeq3o&feature=youtu.be

thomasjacquin, 2019-03-12:
Hey Michael,

You would get this result if you had bin 2 enabled. I have never seen the gain create this kind of problem.

Can you consistently reproduce the behaviour by changing gain to 0?

maphilli14, 2019-03-12:
It was binned!  I've reset to 1x1 and will check, thanks!  Do you know why the resolution would change?  The font is so much sharper in 1x1!!  TYVM

thomasjacquin, 2019-03-12:
In bin 2 mode, the camera groups the signal from 4 adjacent photosites to create 1 pixel on the final image. This results in an image being 4 times small (half width, half height). Basically, you loose resolution but you gain exposure time.
The font is divided by the bin value in order to remain the same size. Let's say that it was 100px tall, it is now 50px tall. It looks the same height but the resolution has been divided by 2.

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys