AllskyTeam/allsky #71: Image not updating images not saved

pterodyne, 2019-03-21:
My allsky is a PI3b with asi174mm.  It seems that my liveview image stop updating.  If I run the script restartCapture.sh from the command line it updates again.  So not sure if the camera is causing a problem? or if something is going on with the scripts.  

I completely removed all traces of the software from the system and reinstalled just now.  After about 30 minutes the same thing started happening.

Thoughts?

thomasjacquin, 2019-03-21:
Try running the program manually from the command line and you may have a stacktrace with more info to identify the issue.
`./allsky.sh`

pterodyne, 2019-03-21:
Didn't help but on further investigation I find this in dmesg:

`[   19.401949] usb 1-1.3: reset high-speed USB device number 5 using dwc_otg
[   19.741948] usb 1-1.3: reset high-speed USB device number 5 using dwc_otg
[   19.872194] usb 1-1.3: device descriptor read/all, error -71
[   19.971938] usb 1-1.3: reset high-speed USB device number 5 using dwc_otg
[   20.071982] usb 1-1.3: device descriptor read/64, error -71
[   20.291989] usb 1-1.3: device descriptor read/64, error -71
[   20.512007] usb 1-1.3: reset high-speed USB device number 5 using dwc_otg
[   20.542885] usb 1-1.3: device descriptor read/8, error -71
[   20.697230] usb 1-1.3: device descriptor read/8, error -71
[   20.911955] usb 1-1.3: reset high-speed USB device number 5 using dwc_otg
[   20.943524] usb 1-1.3: device firmware changed
[   20.943742] usb 1-1.3: USB disconnect, device number 5
[   21.042013] usb 1-1.3: new high-speed USB device number 6 using dwc_otg
[   21.141939] usb 1-1.3: device descriptor read/64, error -71
[   21.361926] usb 1-1.3: device descriptor read/64, error -71
[   21.581957] usb 1-1.3: new high-speed USB device number 7 using dwc_otg
[   21.681930] usb 1-1.3: device descriptor read/64, error -71
[   21.921932] usb 1-1.3: device descriptor read/64, error -71
[   22.052045] usb 1-1-port3: attempt power cycle
[   22.711941] usb 1-1.3: new high-speed USB device number 8 using dwc_otg
[   22.754387] usb 1-1.3: device descriptor read/8, error -71
[   22.904843] usb 1-1.3: device descriptor read/8, error -71
[   23.121933] usb 1-1.3: new high-speed USB device number 9 using dwc_otg
[   23.153130] usb 1-1.3: New USB device found, idVendor=03c3, idProduct=174b, bcdDevice= 0.00
[   23.153143] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   23.153153] usb 1-1.3: Product: ASI174MM
[   23.153162] usb 1-1.3: Manufacturer: ZWO`

I had just been looking at lsusb and the camera was present.  Then once a moment ago I noticed it wasn't, even after several reboots.  Then it was back.  hopefully it's not the camera.  We already had a similar issue a while back and I replaced the cable.  Maybe it's the port on the pi.  It's from the junk drawer after all!

I couldn't find anything on error -71.

My observatory is off-grid so 12V AGM batteries.  The allsky is powered via battery --> 12V dc to 5V dc buck converter --> PI usb.  There are no undervoltage warnings.  Already been through that with earlier iterations. :)

thomasjacquin, 2019-03-21:
As long as your battery system is capable of providing 5V@2.5Amps all the time, you shouldn't have any problem. One use reported that he had to replace the power cable of the Pi for a cable with larger leads on it and he soldered the cable directly to the board because the micro-USB connector wasn't handling the power needs very well.
Yes, try switching for a different RPi to see if it's still happening.

pterodyne, 2019-03-21:
Thanks Ill try a new PI..

AnWeNet, 2019-03-21:
Did you try another USB cable?

pterodyne, 2019-03-22:
I didn't, But the I did replace the cable altogether a couple of months ago (redid the housing and cabling), so in a sense, yes, just not in the current time frame/problem.  The Pi and the camera are the only things that didn't change.  I even replaced the SD card because the prior iteration stopped booting at some point.  I can't do much now as I live 2500 miles away :)   Im having a new Pi, cable and SD card sent to my remote hands person.  Hopefully it isn't the camera.  

I didn't post the SD card problem here because it's not Allsky camera specific.  But something to think about for anyone setting one of these up. Ive gone through several cards over the years.  Not sure, maybe it's my somewhat unreliable off-grid power situation, or the extreme temperature fluctuations going on in my high desert environment.  Basically the PI won't boot, even though the image writes to the SD card fine. Even deleting the partitions and starting from scratch doesn't help.  New SD card solves problem.

pterodyne, 2019-03-29:
I can confirm, in my case we switched the cable to a different USB port and the camera is working again.  It was a "drawer pi" after all (as in a junk drawer with a bunch of other techno scrap).  Watch out for the drawer pi.  I still wonder about the thermal variations.

pterodyne, 2019-04-05:
Sadly it is broken again.  We will try again with a new cable.  But it's looking more like the camera.  Anyone have any thoughts on why this camera would be stopping delivering images after some short period of time?  I might try the camera in windows and just capture images repetitively until it fails or it doesn't.

Seems unlikely to be the Pi unless all the USB ports are on the fritz...

pterodyne, 2019-04-06:
Side note, it shows in lsusb, but the camera isn't working.  Here is the tail of dmesg:

[   21.101928] usb 1-1.5: reset high-speed USB device number 4 using dwc_otg
[  115.042515] random: crng init done
[  115.042528] random: 7 urandom warning(s) missed due to ratelimiting
[  336.858757] usb 1-1.5: reset high-speed USB device number 4 using dwc_otg
[  344.388630] usb 1-1.5: reset high-speed USB device number 4 using dwc_otg
[  361.728254] usb 1-1.5: reset high-speed USB device number 4 using dwc_otg
[  570.953552] usb 1-1.5: reset high-speed USB device number 4 using dwc_otg
[ 8100.007723] usb 1-1.5: USB disconnect, device number 4
[ 8100.308411] usb 1-1.5: new high-speed USB device number 5 using dwc_otg
[ 8100.439429] usb 1-1.5: New USB device found, idVendor=03c3, idProduct=174b, bcdDevice= 0.00
[ 8100.439443] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 8100.439452] usb 1-1.5: Product: ASI174MM
[ 8100.439462] usb 1-1.5: Manufacturer: ZWO
[35743.655807] usb 1-1.5: USB disconnect, device number 5
[35743.965470] usb 1-1.5: new high-speed USB device number 6 using dwc_otg
[35744.096499] usb 1-1.5: New USB device found, idVendor=03c3, idProduct=174b, bcdDevice= 0.00
[35744.096513] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[35744.096522] usb 1-1.5: Product: ASI174MM
[35744.096531] usb 1-1.5: Manufacturer: ZWO
[38097.575876] usb 1-1.5: USB disconnect, device number 6
[38097.875791] usb 1-1.5: new high-speed USB device number 7 using dwc_otg
[38098.006815] usb 1-1.5: New USB device found, idVendor=03c3, idProduct=174b, bcdDevice= 0.00
[38098.006830] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[38098.006839] usb 1-1.5: Product: ASI174MM
[38098.006848] usb 1-1.5: Manufacturer: ZWO
[275522.728544] usb 1-1.5: USB disconnect, device number 7
[275523.021602] usb 1-1.5: new high-speed USB device number 8 using dwc_otg
[275523.152636] usb 1-1.5: New USB device found, idVendor=03c3, idProduct=174b, bcdDevice= 0.00
[275523.152649] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[275523.152659] usb 1-1.5: Product: ASI174MM
[275523.152668] usb 1-1.5: Manufacturer: ZWO

Again, Power is not an issue for me, unless it's not being reported. In the past Ive seen the "undervoltage" errors.  Im using a DC DC converter off of a 12V Battery array, no undervoltage errors.

thomasjacquin, 2019-04-06:
Have you tried the recommended power adapter that comes with the Pi? It's a 2.5A power supply.
They have increased from 2.1A to 2.5A when they released the Pi3 because it takes a bit more power.

I am not too concerned about the voltage in your case but I think there might be a drop in the amount of amps provided to the camera. If the Pi can't provide the required Amps to the camera, I suppose it would make the peripheral stop or restart.

pterodyne, 2019-04-06:
Thanks.  My setup is all off grid, so no AC at all. My DC DC adapter can supply up to 10A

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved and/or this doesn't appear to be an issue with the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys