AllskyTeam/allsky #72: Fresh boot doesn't start service properly

maphilli14, 2019-03-28:
This happens each boot:

Last login: Mon Mar 25 09:06:43 2019 from 192.168.115.2
pi@allsky:~ $ sudo service allsky status
* allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: e
   Active: inactive (dead)
pi@allsky:~ $ uptime
 12:36:32 up 5 min,  2 users,  load average: 0.26, 0.50, 0.27
pi@allsky:~ $ sudo service allsky restart
pi@allsky:~ $ sudo service allsky status
* allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: e
   Active: active (running) since Thu 2019-03-28 12:36:37 EDT; 7s ago
 Main PID: 2530 (allsky.sh)
   CGroup: /system.slice/allsky.service
           |-2530 /bin/bash /home/pi/allsky/allsky.sh
           `-2589 ./capture -autoexposure 1 -autogain 1 -bin 1 -brightness 5 -da

Mar 28 12:36:37 allsky systemd[1]: Started All Sky Camera.

maphilli14, 2019-03-28:
[ 9045.797435] usb 1-1.3: USB disconnect, device number 8
[ 9046.094381] usb 1-1.3: new high-speed USB device number 9 using dwc_otg
[ 9046.235501] usb 1-1.3: New USB device found, idVendor=03c3, idProduct=174b
[ 9046.235518] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 9046.235546] usb 1-1.3: Product: ASI174MM
[ 9046.235554] usb 1-1.3: Manufacturer: ZWO
[11753.254198] usb 1-1.3: reset high-speed USB device number 9 using dwc_otg
[12765.989376] usb 1-1.3: USB disconnect, device number 9
[12766.281278] usb 1-1.3: new high-speed USB device number 10 using dwc_otg
[12766.412234] usb 1-1.3: New USB device found, idVendor=03c3, idProduct=174b
[12766.412249] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[12766.412258] usb 1-1.3: Product: ASI174MM
[12766.412266] usb 1-1.3: Manufacturer: ZWO
[13125.925389] usb 1-1.3: USB disconnect, device number 10
[13126.223848] usb 1-1.3: new high-speed USB device number 11 using dwc_otg
[13126.354718] usb 1-1.3: New USB device found, idVendor=03c3, idProduct=174b
[13126.354735] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[13126.354744] usb 1-1.3: Product: ASI174MM
[13126.354752] usb 1-1.3: Manufacturer: ZWO
[13245.989358] usb 1-1.3: USB disconnect, device number 11
[13246.294476] usb 1-1.3: new high-speed USB device number 12 using dwc_otg
[13246.425553] usb 1-1.3: New USB device found, idVendor=03c3, idProduct=174b
[13246.425570] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[13246.425580] usb 1-1.3: Product: ASI174MM
[13246.425588] usb 1-1.3: Manufacturer: ZWO
[13365.797406] usb 1-1.3: USB disconnect, device number 12
[13366.205319] usb 1-1.3: new high-speed USB device number 13 using dwc_otg
[13366.336170] usb 1-1.3: New USB device found, idVendor=03c3, idProduct=174b
[13366.336186] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[13366.336195] usb 1-1.3: Product: ASI174MM
[13366.336203] usb 1-1.3: Manufacturer: ZWO
[13425.445363] usb 1-1.3: USB disconnect, device number 13
[13425.765625] usb 1-1.3: new high-speed USB device number 14 using dwc_otg
[13425.896547] usb 1-1.3: New USB device found, idVendor=03c3, idProduct=174b
[13425.896563] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[13425.896572] usb 1-1.3: Product: ASI174MM
[13425.896580] usb 1-1.3: Manufacturer: ZWO
[13845.797386] usb 1-1.3: USB disconnect, device number 14
[13846.198240] usb 1-1.3: new high-speed USB device number 15 using dwc_otg
[13846.339139] usb 1-1.3: New USB device found, idVendor=03c3, idProduct=174b
[13846.339154] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[13846.339163] usb 1-1.3: Product: ASI174MM
[13846.339173] usb 1-1.3: Manufacturer: ZWO
[14032.165419] usb 1-1.3: USB disconnect, device number 15
[14032.459448] usb 1-1.3: new high-speed USB device number 16 using dwc_otg
[14032.590370] usb 1-1.3: New USB device found, idVendor=03c3, idProduct=174b
[14032.590387] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[14032.590396] usb 1-1.3: Product: ASI174MM
[14032.590404] usb 1-1.3: Manufacturer: ZWO
[14032.677383] usb 1-1.3: USB disconnect, device number 16
[14033.069353] usb 1-1.3: new high-speed USB device number 17 using dwc_otg
[14033.210447] usb 1-1.3: New USB device found, idVendor=03c3, idProduct=174b
[14033.210482] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[14033.210492] usb 1-1.3: Product: ASI174MM

thomasjacquin, 2019-03-28:
I would try a different USB cable in order to debug the issue.

maphilli14, 2019-03-29:
I will, just had this short, right angle ordered to fit inside the tube so I hope that's not it...  in the meantime, I'm going to put in a cron to restart about ~1hr. Also I wrote a small script to animate the last few 10's of minutes... https://f002.backblazeb2.com/file/allsky/LiveRefresh.html

maphilli14, 2019-04-01:
I moved the power supply from Android adapter to PoE and while it still gives me undervolatge, I'm good for the past  few days:

[117799.340781] Voltage normalised (0x00000000)
[117836.781011] Under-voltage detected! (0x00050005)
[117840.940967] Voltage normalised (0x00000000)
[118096.782236] rpi_firmware_get_throttled: 7 callbacks suppressed
[118096.782245] Under-voltage detected! (0x00050005)
[118103.022202] rpi_firmware_get_throttled: 7 callbacks suppressed
[118103.022207] Voltage normalised (0x00000000)
[118165.422577] Under-voltage detected! (0x00050005)
[118169.582640] Voltage normalised (0x00000000)
[118196.622733] Under-voltage detected! (0x00050005)
[118200.782658] Voltage normalised (0x00000000)
[118435.823816] rpi_firmware_get_throttled: 5 callbacks suppressed
[118435.823825] Under-voltage detected! (0x00050005)
[118446.223834] rpi_firmware_get_throttled: 5 callbacks suppressed
[118446.223840] Voltage normalised (0x00000000)
[118454.543896] Under-voltage detected! (0x00050005)
[118475.343955] Voltage normalised (0x00000000)
[118477.424014] Under-voltage detected! (0x00050005)
[118481.583963] Voltage normalised (0x00000000)
[118905.906043] rpi_firmware_get_throttled: 7 callbacks suppressed
[118905.906047] Voltage normalised (0x00000000)
[118907.986071] rpi_firmware_get_throttled: 8 callbacks suppressed
[118907.986079] Under-voltage detected! (0x00050005)
[118984.946360] Voltage normalised (0x00000000)
[118987.026443] Under-voltage detected! (0x00050005)
[119020.306641] Voltage normalised (0x00000000)
[119022.386632] Under-voltage detected! (0x00050005)
[119286.547676] rpi_firmware_get_throttled: 1 callbacks suppressed
[119286.547680] Voltage normalised (0x00000000)
[119290.707857] rpi_firmware_get_throttled: 1 callbacks suppressed
[119290.707866] Under-voltage detected! (0x00050005)
[119340.627934] Voltage normalised (0x00000000)
[119342.708207] Under-voltage detected! (0x00050005)
[119507.028636] Voltage normalised (0x00000000)
[119511.188721] Under-voltage detected! (0x00050005)
[119918.870591] Voltage normalised (0x00000000)
[119927.190536] Under-voltage detected! (0x00050005)
[120039.511010] Voltage normalised (0x00000000)
[120041.591127] Under-voltage detected! (0x00050005)
[120066.551143] Voltage normalised (0x00000000)
[120068.641291] Under-voltage detected! (0x00050005)
[120228.791700] rpi_firmware_get_throttled: 1 callbacks suppressed
[120228.791706] Voltage normalised (0x00000000)
[120230.871847] rpi_firmware_get_throttled: 1 callbacks suppressed
[120230.871857] Under-voltage detected! (0x00050005)
[120341.112258] Voltage normalised (0x00000000)
[120343.192223] Under-voltage detected! (0x00050005)
[121010.874873] Voltage normalised (0x00000000)
[121015.034970] Under-voltage detected! (0x00050005)
[121243.835704] Voltage normalised (0x00000000)
[121245.915788] Under-voltage detected! (0x00050005)
[121295.836012] Voltage normalised (0x00000000)
[121347.836259] Under-voltage detected! (0x00050005)
[121354.076276] Voltage normalised (0x00000000)
[121466.396743] Under-voltage detected! (0x00050005)
[121470.556810] Voltage normalised (0x00000000)
[121528.796987] Under-voltage detected! (0x00050005)
[121532.956963] Voltage normalised (0x00000000)
[122188.159740] Under-voltage detected! (0x00050005)
[122194.399566] Voltage normalised (0x00000000)
[122664.481140] Under-voltage detected! (0x00050005)
[122670.721090] Voltage normalised (0x00000000)
[123565.124176] Under-voltage detected! (0x00050005)
[123569.284100] Voltage normalised (0x00000000)
[125551.530150] Under-voltage detected! (0x00050005)
[125557.770195] Voltage normalised (0x00000000)
[125663.850657] Under-voltage detected! (0x00050005)
[125670.090562] Voltage normalised (0x00000000)
[126857.775067] Under-voltage detected! (0x00050005)
[126861.935067] Voltage normalised (0x00000000)
[127165.616298] Under-voltage detected! (0x00050005)
[127169.776245] Voltage normalised (0x00000000)
[129757.307673] Under-voltage detected! (0x00050005)
[129763.547759] Voltage normalised (0x00000000)
[131920.518436] Under-voltage detected! (0x00050005)
[131924.678377] Voltage normalised (0x00000000)
[133721.808475] Under-voltage detected! (0x00050005)
[133725.968481] Voltage normalised (0x00000000)
[135344.217382] Under-voltage detected! (0x00050005)
[135348.377294] Voltage normalised (0x00000000)
[138526.635031] Under-voltage detected! (0x00050005)
[138530.795056] Voltage normalised (0x00000000)
[139547.920667] Under-voltage detected! (0x00050005)
[139552.080682] Voltage normalised (0x00000000)
[140060.663443] usb 1-1.3: reset high-speed USB device number 4 using dwc_otg

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys