AllskyTeam/allsky #73: Error during installation

sifank, 2019-04-05:
[allskyInstall.txt](https://github.com/thomasjacquin/allsky/files/3049066/allskyInstall.txt)
I just downloaded and ran the install today (4/5/19).  I'm using Raspbian 9, RPI 3 B.  At the end of running the install.sh script it complains that target 'capture' failed.  Lot's of errors, the first of which is: 
capture.cpp:49:22: error: no matching function for call to ‘cv::Mat::Mat(IplImage*&)’
  cv::Mat mat_img(img );
(with the final ')' in red and an carrot pointing at it.)

I tried this with both the git download and the zip download.

I attached the install output.

maphilli14, 2019-04-08:
where are you seeing output of capture failed?  I just did a fresh OS and app instal yesterday but am finding a broken jpg in place of the live views and don't know why.  oacapture works fine but not allsky :(

sifank, 2019-04-08:
I did a few fresh downloads to a RPI 3B+, Raspian v9 (stretch), and ran the install.sh.  As you see in the original attachment, it failed compiling capture.cpp, errors out on the cv line 49.  From there on, there are lots of other errors (see that attachment).  I've tried reinstalling all the prerequisites and still get the same errors.  I can also just run the 'make all' and get the same.

Could it be a wrong version of opencv?

maphilli14, 2019-04-08:
libopencv*??  I'm on 2.4.9.1 and 4.14.98-v7+ raspbian

thomasjacquin, 2019-04-08:
There seems to be an error with opencv indeed. I haven't tried the latest release of Raspbian yet so I can't confirm what's going on for sure. Can you provide the version of opencv you're using?

sifank, 2019-04-09:
Aloha Thomas,  (just got back home).  Mahalo for your responses.  I did
some investigating and a couple of tests.

The rpi system I was using as a test bed had opencv 4.0.0 install.  So I
went to one of my other rpi's, it's opencv was 2.4.9.1, so I did the allsky
install on it and it works great!

I did some research and it seems like opencv version 3.x and up have a
number of non-backward compatible changes - a tad messy.

This works - sorry for the trouble and mahalo for the help!!

Aloha,

Sifan
Have (gravel) road, will travel ...

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys