AllskyTeam/allsky #74: FTP Issue

spacemanbobby, 2019-04-10:
So I have put the tags into the scripts to upload the files with that has been mentioned... Image-resize.jpg, startrails, keograms and the daily mp4 all get uploaded to the website.... but it is as if it only puts them in the base directory that I set the FTP account to log into. 

lets say... 
Allsky@website.com logs into ftp.website.com
On my site I set this account to access public_html\allsky\
I have the other folders under allsky.... images, keograms, startrails, videos.... all lower caps and they are 755 under the allsky folder. 
When the scripts fire off they all just put them in the base allsky folder even though I have them as \allsky\ for images, \allsky\videos\ for videos... etc... 

I have tried all sorts of combo's on the paths but it always just gets put into the base allsky directory that the ftp account points to for signing into a certain directory. 
I even tried setting the ftp account to sign into \public_html\ and then let the upload scripts try to get to allsky, but it just dumped it in public_html. 

Any ideas. 
At this point I am going into ftp from my computer and moving the files to the respective folders. 

I even bought a second respberry pi board and installed the OS and the Allsky software onto it. Same results once I get it to upload to the ftp account.

thomasjacquin, 2019-04-11:
I believe this is an issue with some web hosts. However you can fix the issue by adding a `mv` command to the lftp command.

`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" "set net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION; mv $FILENAME-resize.$EXTENSION allsky/$FILENAME-resize.$EXTENSION; bye"`

This will move the uploaded image to the allsky directory

You will have to update the lftp command in a few scripts in order to get it to work for startrails, keograms and timelapse as well.

gremlinforce007, 2019-04-13:
Thanks Thomas this was super helpful :)

spacemanbobby, 2019-04-13:
Thanks for the additional information, the images resize are now loading in
the images folder... Just got to see if the other stuff starts working
tonight and tomorrow morning when it finishes.

gremlinforce007, 2019-04-13:
@spacemanbobby can you please post the script edits you've done for timelapses, keograms and startrails please ?

thomasjacquin, 2019-04-14:
These should work but I haven't tested them. Let me know if they don't and I'll update them.

SaveImageDay.sh and saveImageNight.sh:
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set net:max-retries 1; set net:timeout 20; put $FILENAME-resize.$EXTENSION;  mv $FILENAME-resize.$EXTENSION; $IMGDIR/$FILENAME-resize.$EXTENSION; bye"`

Timelapse:
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST" -e "set net:max-retries 1; put images/$1/allsky-$1.mp4; mv allsky-$1.mp4 $MP4DIR/allsky-$1.mp4; bye"`

Keogram (in endOfNight.sh):
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST"  -e "set net:max-retries 1; put $OUTPUT; mv $OUTPUT $KEOGRAM_DIR/$OUTPUT; bye"`

Startrails (in endOfNight.sh):
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST" -e "set net:max-retries 1; put $OUTPUT; mv $OUTPUT $STARTRAILS_DIR/$OUTPUT; bye"`

gremlinforce007, 2019-04-14:
Thanks i made an edit i'll let you know tonight :)

gremlinforce007, 2019-04-14:
Hi @thomasjacquin 

Just tried your post: Still went to root folder:

See code below:

Timelapse:
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$MP4DIR" -e "set net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no;put images/$1/allsky-$1.mp4; mv allsky-$1.mp4 $allsky/videos/allsky-$1.mp4; bye"`

KeoGram:
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$KEOGRAM_DIR" \
                        -e "set net:max-retries 1;set net:timeout 20; set ftp:ssl-allow no; put $OUTPUT; mv $OUTPUT $allsky/keogram/$OUTPUT; bye"`

StarTrails:
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$STARTRAILS_DIR" \
			-e "set net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put $OUTPUT; mv $OUTPUT $allsky/startrails/$OUTPUT;  bye"`

Question should there be a `$` near >> $allsky/startrails/$OUTPUT; ?
[image]

spacemanbobby, 2019-04-15:
Been busy this weekend... Will update the LFTP scripts with what Thomas
said for the startrails, keograms and videos tuesday.. The image-resize.jpg
goes into my images folder now.

I had to modify my ftp login script since the ftp account already logged
into the base of allsky... just removed allsky from the paths so the mv
command would just move it from allsky base to the specified folders.. I'll
post that also on tuesday.

thomasjacquin, 2019-04-15:
@gremlinforce007 Looking at your Cpanel structure, I think you'll need to remove that dollar sign in front of `allsky` and it should just work.

gremlinforce007, 2019-04-16:
So removing the dollar sign worked for timelapse .. not startrails or keogram any ideas ?

KeoGram:
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$KEOGRAM_DIR" \ -e "set net:max-retries 1;set net:timeout 20; set ftp:ssl-allow no; put $OUTPUT; mv $OUTPUT $allsky/keogram/$OUTPUT; bye"`

StarTrails:
`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$STARTRAILS_DIR" \ -e "set net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put $OUTPUT; mv $OUTPUT $allsky/startrails/$OUTPUT; bye"`

thomasjacquin, 2019-04-16:
Can you take a screenshot of your cpanel with the content of the allsky directory? 
Also I think the directory is called keograms on my server. If these directories don’t exist, you’ll have to create them.

gremlinforce007, 2019-04-16:
> Yep added the s to keogram 

[image]

thomasjacquin, 2019-04-16:
If it still doesn't send, you can try calling `scripts/endOfNight.sh` manually from the terminal and look for errors.

spacemanbobby, 2019-04-16:
Ok.. so here is what I have for settings on my scripts and ftp settings...
blanked out my site and password.....

resize-image gets uploaded to the images directory now. the mp4 file gets
uploaded to the videos folder now. But keograms and startrails still get
dumped in the allsky base directory of my ftp site. I have put the error
codes from running the endOfNight.sh script at the bottom. See if you can
find something I missed.

# FTP/SFTP settings
PROTOCOL='ftp'
USER='allsky@webpage.com'
PASSWORD='*******'
HOST='ftp.webpage.com'
IMGDIR='/images'
MP4DIR='/videos'
KEOGRAM_DIR='/keograms'
STARTRAILS_DIR='/startrails'

Reason I changed paths is that the FTP account I created in CPanel, logs
directly into the allsky

saveImageDay.sh

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set
net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put
$FILENAME-resize.$EXTENSION; mv $FILENAME-resize.$EXTENSION
images/$FILENAME-resize.$EXTENSION; bye"

saveImageNight.sh

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set
net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put
$FILENAME-resize.$EXTENSION; mv $FILENAME-resize.$EXTENSION
images/$FILENAME-resize.$EXTENSION; bye"

timelapse.sh

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$MP4DIR" -e "set
net:max-retries 1; set ftp:ssl-allow no; put images/$1/allsky-$1.mp4; mv
allsky-$1.mp4 $MP4DIR/allsky-$1.mp4; bye"

endOfNight.sh

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$KEOGRAM_DIR" -e "set
net:max-retries 1; set ftp:ssl-allow no; put $OUTPUT; mv $OUTPUT
$KEOGRAM_DIR/$OUTPUT; bye"

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$STARTRAILS_DIR" -e "set
net:max-retries 1; set ftp:ssl-allow no; put $OUTPUT; mv $OUTPUT
$STARTRAILS_DIR/$OUTPUT; bye"

622803 bytes transferred in 1 second (536.7 KiB/s)
mv: Access failed: 550 Sorry, but that file doesn't exist
(/home/pi/allsky/images/20190416/startrails/startrails-20190416.jpg)

399530 bytes transferred in 1 second (328.1 KiB/s)
mv: Access failed: 550 Sorry, but that file doesn't exist
(/home/pi/allsky/images/20190416/keogram/keogram-20190416.jpg)

13018800 bytes transferred in 7 seconds (1.82 MiB/s)
mv allsky-20190416.mp4=>/videos/allsky-20190416.mp4 [Waiting for
response...]
rename successful

thomasjacquin, 2019-04-16:
Ah yes, I see, it's because `$OUTPUT` evaluates to `home/pi/allsky/images/20190416/keogram/keogram-20190416.jpg` instead of `keogram-20190416.jpg` instead of `keogram-20190416.jpg`.

Adding a variable like this should work I think.
OUTPUT="/home/pi/allsky/images/$LAST_NIGHT/keogram/keogram-$LAST_NIGHT.jpg"
KEOGRAM="keogram-$LAST_NIGHT.jpg"

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$KEOGRAM_DIR" \
                        -e "set net:max-retries 1; put $OUTPUT; mv $KEOGRAM
$KEOGRAM_DIR/$KEOGRAM; bye"

spacemanbobby, 2019-04-16:
Would the same thing be done for the startrails error. that is in the email
that you saw this.

thomasjacquin, 2019-04-16:
Yes same idea for startrails, just replace `KEOGRAM` by `STARTRAILS`

spacemanbobby, 2019-04-16:
I made the changes and ran the endofnight script.. Everything uploads to
the correct directories now.

Thanks for helping identify these things. I'll post my settings so other
user who have webpages like mine may see what needs to be done.

gremlinforce007, 2019-04-16:
@spacemanbobby Thanks for your assistance as well, can you kindly post your scripts as i'm still having issues :)

spacemanbobby, 2019-04-17:
One thing I need to ask.. Did you create an FTP account on your website? If
so.... what directory does it default log into. As I said... Mine logs into
my allsky. If your's does the same then maybe the scripts I modified will
work... I'll post them tonight... Have to find them first.

spacemanbobby, 2019-04-17:
OK.. here is a cut and paste of all the things I changed... and now it
works. So if you want.. cut and paste some of the things I pasted from my
scripts over your's. But make a backup copy just in case of the originals.

On my website I created under the allsky directory. 4 folders.. images,
videos, keograms, and startrails Make sure your allsky has 755 as chmod
settings and the folders under it should inherit those settings.

ftp-settings.sh

# FTP/SFTP settings
PROTOCOL='ftp'
USER='allsky@webpage.com'
PASSWORD='*******'
HOST='ftp.webpage.com'
IMGDIR='/images'
MP4DIR='/videos'
KEOGRAM_DIR='/keograms'
STARTRAILS_DIR='/startrails'

Reason I changed paths is that the FTP account I created in CPanel, logs
directly into the allsky directory.

saveImageDay.sh

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set
net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put
$FILENAME-resize.$EXTENSION; mv $FILENAME-resize.$EXTENSION
images/$FILENAME-resize.$EXTENSION; bye" &

saveImageNight.sh

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set
net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put
$FILENAME-resize.$EXTENSION; mv $FILENAME-resize.$EXTENSION
images/$FILENAME-resize.$EXTENSION; bye"

timelapse.sh

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$MP4DIR" -e "set
net:max-retries 1; set ftp:ssl-allow no; put images/$1/allsky-$1.mp4; mv
allsky-$1.mp4 $MP4DIR/allsky-$1.mp4; bye"

endOfNight.sh

OUTPUT="/home/pi/allsky/images/$LAST_NIGHT/keogram/keogram-$LAST_NIGHT.jpg"
                KEOGRAM="keogram-$LAST_NIGHT.jpg"
lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$KEOGRAM_DIR" -e "set
net:max-retries 5; set ftp:ssl-allow no; put $OUTPUT; mv $KEOGRAM
$KEOGRAM_DIR/$KEOGRAM; bye"

OUTPUT="/home/pi/allsky/images/$LAST_NIGHT/startrails/startrails-$LAST_NIGHT.jpg"
                STARTRAILS="startrails-$LAST_NIGHT.jpg"
lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$STARTRAILS_DIR" -e "set
net:max-retries 5; set ftp:ssl-allow no; put $OUTPUT; mv $STARTRAILS
$STARTRAILS_DIR/$STARTRAILS; bye"

Let your system run through the night and when it is done in the morning,
see if it did it's thing.. With these settings and Thomas's help everything
now loads to the correct directories. Before  he helped, it was just
uploading to my base allsky directory. Reason I say wait, is that the
system has to generate the keograms, startrails and the video(mp4) from the
images that it takes on the Raspberry Pi device... Then when the
endOfNight.sh script runs... it gathers up all the images one group at a
time and creates the jpg's and mp4 that need to be uploaded via ftp.

gremlinforce007, 2019-04-17:
Hi Team, 

My Startrails and keograms are still not working:

# FTP/SFTP settings
PROTOCOL='ftp'
USER='allsky@webpage.com'
PASSWORD='*******'
HOST='ftp.webpage.com'
IMGDIR='/images'
MP4DIR='/videos'
KEOGRAM_DIR='/keograms'
STARTRAILS_DIR='/startrails'

**> scripts/endOfNight.sh: line 22: keogram-20190417.jpg: command not found**

[endOfNight.txt](https://github.com/thomasjacquin/allsky/files/3091947/endOfNight.txt)

Can you let me know whats wrong ?

thomasjacquin, 2019-04-18:
Try getting rid of the space in `KEOGRAM= "keogram-$LAST_NIGHT.jpg`

gremlinforce007, 2019-04-18:
Thanks that seemed to work, now i'm getting this on line 24:

**scripts/endOfNight.sh: line 24: -e: command not found**

gremlinforce007, 2019-04-18:
**FIXED YAY 

**

ZebrasHaveBigEars, 2019-05-10:
ONLY IF YOU HAVE MAJOR FTP PROBLEMS, test your ftp using this simple code saved into test.sh and then run from terminal on your pi from within the directory using ./test.sh. Subcommand "debug" provides lots of help for errors and "dir" tells you the starting path for your allsky directory.

#!/bin/bash

lftp "
set ssl:verify-certificate no
set net:max-retries 1
set net:timeout 20
open ftp://website.com
user 'username' 'password'
debug
cd
dir
lpwd
bye"
echo "FTP DONE"
date

In fact, I substituted this lftp code into the allsky scripts and simply added the transfer code provided by Thomas into it. After a week of trying to get the original code to work or using more complex code, this simple code was the only way I managed to get it to work and it was very easy to do. The subcommand "set ssl:verify-certificate/ftp.website.com no" did not work for me for some reason.

It should be easy to substitute in the new code for Thomas' slicker code but which is harder to troubleshoot:

1. Make a copy of the original code
2. Put # in front of the original lftp command line
3. Once tested copy in the new lftp code below the original lftp line (not the first line)
4. Put in new code using "cd folder/folder" etc to move to the correct folder in your remote website
5. Check Thomas' original lftp code and put in the appropriate "put" command into the script
6. Make sure you indent the new code to be positioned in the same place as the original
7. Make sure you don't forget the fi command!
8. Save. Test using ./allsky.sh

If anyone wants examples then let me know.

elineberry75, 2019-10-27:
Forgive my ignorance but while I'm pretty good at most "techy" things, I'm pretty new to Pi's and Linux.  I have the pi up and running with Allsky and an ASI20MC-S and it seems to work great!  I can access it via GUI and see the live images but I'm having problems with getting my FTP to work correctly as NOTHING is uploading.  I've ran the scripts manually to look for errors and I get the following error:  Fatal error: Certificate verification: certificate common name doesn't match requested host name......
I've edited the scripts to use "set ftp:ssl-allow no" like I've read in some of your examples on these posts but it still does not work.  I'm using GoDaddy hosting and configured FTP in the cPanel.  I can get it to connect without problems in Filezilla using the same credentials I have configured in ftp-setting.sh.  Any suggestions??  I think I could figure out other small issues to get it working like it should if I could ever get the FTP to communicate.

Also, on a side note, does sameImageDay(or Night) upload every image as a separate file or does it overwrite it each time to the filename I designated in settings.json?  My image name in settings.json is simply "image.jpg" and it is in the allsky main directory which is where my FTP settings are setup to deposit the image.  In other words, I do not have a separate directory for images like I do for startrails, MP4, keograms, etc.  Will that work or do I need a "images" directory?

Thanks in advance for any help!!

spacemanbobby, 2019-10-27:
I'll see if I can send you what I have... I had almost the exact same
issue.. Once I got it to connect it wouldn't put things in the right
folders... I am busy right now but will look at my email history and send
what changes I made.

elineberry75, 2019-10-27:
Thank you.  That would be great!  I really appreciate it.

elineberry75, 2019-10-27:
So, I've been working on it some and although I'm still getting the "Certificate not matching" error, it is uploading the files.  The image and video files are being deposited in the correct directories, however the keograms and startrails are not.  Like you said, they are going into the allsky root directory.  My ftp_settings.sh file is below.  Permissions for all of the folders are the same (0755).

#!/bin/bash

# FTP/SFTP settings
PROTOCOL='ftp'
USER='xxxxxxxxx'
PASSWORD='xxxxxxxx'
HOST='xxxxxxxxxxxxxxxxxx'
IMGDIR='/'
MP4DIR='/videos'
KEOGRAM_DIR='/keograms'
STARTRAILS_DIR='/startrails'

spacemanbobby, 2019-10-27:
OK.. here is a cut and paste of all the things I changed... and now it
works. So if you want.. cut and paste some of the things I pasted from my
scripts over your's. But make a backup copy just in case of the originals.

On my website I created under the allsky directory. 4 folders.. images,
videos, keograms, and startrails Make sure your allsky has 755 as chmod
settings and the folders under it should inherit those settings.

ftp-settings.sh

# FTP/SFTP settings
PROTOCOL='ftp'
USER='allsky@webpage.com'
PASSWORD='*******'
HOST='ftp.webpage.com'
IMGDIR='/images'
MP4DIR='/videos'
KEOGRAM_DIR='/keograms'
STARTRAILS_DIR='/startrails'

Reason I changed paths is that the FTP account I created in CPanel, logs
directly into the allsky directory.

saveImageDay.sh

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set
net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put
$FILENAME-resize.$EXTENSION; mv $FILENAME-resize.$EXTENSION
images/$FILENAME-resize.$EXTENSION; bye" &

saveImageNight.sh

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set
net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put
$FILENAME-resize.$EXTENSION; mv $FILENAME-resize.$EXTENSION
images/$FILENAME-resize.$EXTENSION; bye"

timelapse.sh

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$MP4DIR" -e "set
net:max-retries 1; set ftp:ssl-allow no; put images/$1/allsky-$1.mp4; mv
allsky-$1.mp4 $MP4DIR/allsky-$1.mp4; bye"

endOfNight.sh

                OUTPUT="/home/pi/allsky
/images/$LAST_NIGHT/keogram/keogram-$LAST_NIGHT.jpg"
                KEOGRAM="keogram-$LAST_NIGHT.jpg"
lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$KEOGRAM_DIR" -e "set
net:max-retries 5; set ftp:ssl-allow no; put $OUTPUT; mv $KEOGRAM
$KEOGRAM_DIR/$KEOGRAM; bye"

                OUTPUT="/home/pi/allsky
/images/$LAST_NIGHT/startrails/startrails-$LAST_NIGHT.jpg"
                STARTRAILS="startrails-$LAST_NIGHT.jpg"
lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$STARTRAILS_DIR" -e "set
net:max-retries 5; set ftp:ssl-allow no; put $OUTPUT; mv $STARTRAILS
$STARTRAILS_DIR/$STARTRAILS; bye"

Let your system run through the night and when it is done in the morning,
see if it did it's thing.. With these settings and Thomas's help everything
now loads to the correct directories. Before  he helped, it was just
uploading to my base allsky directory. Reason I say wait, is that the
system has to generate the keograms, startrails and the video(mp4) from the
images that it takes on the Raspberry Pi device... Then when the
endOfNight.sh script runs... it gathers up all the images one group at a
time and creates the jpg's and mp4 that need to be uploaded via ftp.

elineberry75, 2019-10-27:
For some reason, the images and videos are correct but the keograms and startrails are still going in the root directory.  My files are below.  I will let it run through the night and check it tomorrow but when I manually run the scripts, it still puts the startrails and keograms in the root directory.  Not sure what is different for those and not the images and videos which seems to be working correctly.

saveImageDay.sh
lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put $FILENAME-resize.$EXTENSION; mv $FILENAME-resize.$EXTENSION images/$FILENAME-resize.$EXTENSION; bye" &

saveImageNight.sh
lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put $FILENAME-resize.$EXTENSION; mv $FILENAME-resize.$EXTENSION images/$FILENAME-resize.$EXTENSION; bye" &

timelapse.sh
lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$MP4DIR" -e "set net:max-retries 1; set ftp:ssl-allow no; put images/$1/allsky-$1.mp4; mv allsky-$1.mp4 $MP4DIR/allsky-$1.mp4; bye"

endOfNight.sh
lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$KEOGRAM_DIR" -e "set net:max-retries 5; set ftp:ssl-allow no; put $OUTPUT; mv $KEOGRAM $KEOGRAM_DIR/$KEOGRAM; bye"

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$STARTRAILS_DIR" -e "set net:max-retries 5; set ftp:ssl-allow no; put $OUTPUT; mv $STARTRAILS $STARTRAILS_DIR/$STARTRAILS; bye"

elineberry75, 2019-10-28:
So, I figured it out with the help of spacemanbobby. Thanks!  The problem was with the "mv" command in endOfNight.sh.  Here's what I ended up with that worked:

endOfNight.sh
lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$KEOGRAM_DIR" -e "set net:max-retries 5; set ftp:ssl-allow no; put $OUTPUT; mv keogram-$LAST_NIGHT.jpg $KEOGRAM_DIR/keogram-$LAST_NIGHT.jpg; bye"

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$STARTRAILS_DIR" -e "set net:max-retries 5; set ftp:ssl-allow no; put $OUTPUT; mv startrails-$LAST_NIGHT.jpg $STARTRAILS_DIR/startrails-$LAST_NIGHT.jpg; bye" 

The only thing I see having an error at this point and maybe it doesn't matter is the postData.sh script that is called on at the very beginning of endOfNight.sh.  For whatever reason, it does not send the data.json file like it is supposed to.

spacemanbobby, 2019-10-29:
Glad I was able to help.

maserowik, 2020-08-06:
Sorry to bring up a old thread but

> The only thing I see having an error at this point and maybe it doesn't matter is the postData.sh script that is called on at the very beginning of endOfNight.sh. For whatever reason, it does not send the data.json file like it is supposed to.

on my postData.sh I added to following and the data.json appears to get updated.

lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put data.json; mv data.json data.json;  bye"

spacemanbobby, 2020-08-07:
When you installed the ne version, did you need to modify the .SH files to
get the FTP to work. Other than entering your FTP information into the FTP
Upload file.

Let me know when you can, thanks.

maserowik, 2020-08-07:
> When you installed the ne version, did you need to modify the .SH files to get the FTP to work. Other than entering your FTP information into the FTP Upload file. Let me know when you can, thanks.
> […](#)

Still fighting with setting up the FTP.  I followed what was post here but no luck.  I can see the files uploaded to my hosting site but they are not moving to the correct directory.  The same for the resized image the files are there but not showing up on my website.

If you like to go off list I cam be reached at mserowik@gmail.com

Mike

thomasjacquin, 2020-08-07:
@maserowik What's your website address?

spacemanbobby, 2020-08-07:
My Website is Spacemanbob.com

/home2/spacemb6/public_html/allsksy  <- That is where my Images land... It
was all working on my Pi 3 board before it died... I have added the lftp
stuff for no ssl... etc.. but now nothing is loading through FTP now.

I know I added an s in allsksy. I was testing things a while back and just
kept to that directory.

maserowik, 2020-08-07:
> @maserowik What's your website address?

n3bsq.com/allsky

maserowik, 2020-08-07:
[n3bsq.com.docx](https://github.com/thomasjacquin/allsky/files/5043974/n3bsq.com.docx)

thomasjacquin, 2020-08-08:
@maserowik 
In config.js change this line
`imageName: "domains/n3bsq.com/public_html/allsky/image-resize.jpg"`
for this one
`imageName: "image-resize.jpg"`

thomasjacquin, 2020-08-08:
@spacemanbobby What's the error in the terminal when you launch `./allsky.sh`? Did you set `UPLOAD_IMG=true` in `config.sh`?

maserowik, 2020-08-08:
With much support for the forum and other user,  There are too many to list.  The allsky website at n3bsq.com/allsky appears to be working with a few issues.
when hitting the n3bsq.com/allsky, O get a aurora forecast stating it is not dark yet come back in 00 minutes.  not sure how to fix that.  Also on the time lapse i see the files there but for some reason unplayable on two of my laptops and one of my desktop.

also for the ssh/ssl issue i tried to ad the command to the *.sh files but still not luck.  I followed the following

https://www.librebyte.net/en/ftp/lftp-fatal-error-certificate-verification-not-trusted/
 
I added it to the  /etc/lftp.conf  file.

one step closer.

spacemanbobby, 2020-08-09:
Where is config.js located at.

spacemanbobby, 2020-08-09:
I got it all working.. Was for some reason not liking the old password I
had forever.. When I changed it to a new one.. ftp started.. And the
config.js.. I had that figured out when I went to my webpage and modified
the location to send the images of the day.. So at this point I'm good.

maserowik, 2020-08-10:
When I use FileZilla I land in the following

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

maserowik, 2020-08-10:
OK I want to thank again all who assisted me with my allsky project.   I done so many changes i am not quite sure i can duplicate what I did.  One more was completed today.   I was stuck on the loading image on my live view.  I changed the path in the config.js and nothing worked.  I finally went and cleared my cache on my machine and went to my webhoster and cleared  the website cache.  again not sure if that did anything but now i can see my live image on the website.

n3bsq.com/allsky now appears to be working correctly.

spacemanbobby, 2020-08-10:
Good Job...
Mine seems to be working but I have a issue that I didn't have in my Pi3
version.. The camera posts garbled images every so often.. I have the
option to delete bad images turned on but it didn't do that on my video
upload for the day. Also... is there a way to turn off image stretch
without having to take my unit all apart from the box I have it in to get
to that file. I suspect it is the Json that was moved to the etc folder..
The allsky doesn't see it now that it is moved.

spacemanbobby, 2020-08-18:
Is anyone else getting this type of effect during the capture during the
day.
http://spacemanbob.com/allsksy/videos/allsky-20200817.mp4
I have auto gain set to no and auto exposure set to no. Gain is set to 152.
Exposure is set to 33 seconds.
I have tried all sorts of combo's.. Still getting these flashes... I am
working on the Dark issue also.. Had to have the Dark setting turned on for
5 minutes before I turned it off to get something other than pixels all
over the place on the image.

asi185mc is my camera.

thomasjacquin, 2020-08-18:
I have had reports about this kind of behaviour. I believe it is caused by the auto-exposure calculation. It is not an allsky thing and it seems to affect only certain models. I think it's coming from the ZWO library. Since this is not related to FTP issues, can you submit a new issue report for this? Thanks

weatherTai, 2020-08-25:
> I made the changes and ran the endofnight script.. Everything uploads to the correct directories now.

Hi, I am having the same problem, can not move the files (images, timelapse..) to my /home/weatherv/public_html/allsky/images ....
I tried to folliow this topic but not be successful yet. Ciuld you please share your steps to fix? Big thanks

ebeyonder, 2020-10-05:
I would be most grateful for any help, as I don't know much about Linux, Pi or FTP... 

I managed to set up my own allsky camera here in Singapore. I can see the live image over VNC very well. 

I next tried to set up a website using Thomas' template. The problem is that the FTP uploads do not work.

This is what I have done: 
1. Bought a webhosting plan and domain, 
2. I loaded all the files from the allsky website repo on Github into a 'allsky' directory on my site,
3. Using cPanel, I set up a FTP account, 
4. I went to the ftp-settings.sh script in allsky/scripts and edited the file to include all my FTP details. 
5. I also changed the config.sh file to specify uploads to website as TRUE.

Then... nothing happens. There is no 'live' picture on the website. 

I tried to execute the script by clicking on it and selecting 'execute' and also 'execute in terminal'... but nothing happens. Am I supposed to see anything? There is not even a message saying it failed.

Any advice much appreciated.

EricClaeys, 2021-09-05:
This issue is from an older release of the AllSky software and the issues appear to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks..

closed 2021-09-05 by EricClaeys

orangemaze, 2021-12-15:
Is there an idiot's guide for the current version? I apparently have no idea what I am doing. I have read the thread, and I see no images in any directory. I do see images via http://allsky.local/index.php?page=live_view, but they are not saving. And yes it is currently nighttime, and the date/time are set correctly.

EricClaeys, 2021-12-15:
Test