AllskyTeam/allsky #75: Images not displaying on Website

gremlinforce007, 2019-04-13:
Hi Thomas, 

I'm a newbie, i've built my allsky and see that images are being saved to the rasPi 3+ 
I also see that my files are being uploaded to the directory after I took into account your comments on another thread and modified my night script:

`lftp "$PROTOCOL"://"$USER":"$PASSWORD"@"$HOST":"$IMGDIR" -e "set net:max-retries 1; set net:timeout 20; set ftp:ssl-allow no; put $FILENAME-resize.$EXTENSION;mv $FILENAME-resize.$EXTENSION allsky/$FILENAME-resize.$EXTENSION; bye"`

However when i login into my FTP, i see the file listed as "image-resize.jpg" however this does not display on my website ?

Any thoughts or suggestions ?

thomasjacquin, 2019-04-14:
Hi, so you're seeing image-resize.jpg on your server using an FTP connection. That's a good start!

Now if you navigate to your website, www.yourwebsite.com/allsky/image-resize.jpg, do you see anything?
Send me your website URL and I'll take a look.
Thomas

gremlinforce007, 2019-04-14:
Hey Mate, 

Managed to get it working, changed the file name. loading.jpg in the index.html to "Image-resize" now it works .. unless i've done something wrong

thomasjacquin, 2019-04-14:
That would work. Ideally, you can just set the file name in config.js though.

gremlinforce007, 2019-04-18:
Fixed :) Thanks
[endOfNight sample.txt](https://github.com/thomasjacquin/allsky/files/3094973/endOfNight.sample.txt)

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software and the issue appears to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-05 by EricClaeys