AllskyTeam/allsky #83: Camera exposure

Kilis71, 2019-04-15:
Hi Thomas, 
I would like to know if it is possible to set the camera so that during the day it has the exposure time automatically and in the evening it always automatically sets itself to a preset time. 
Thanks, bye

thomasjacquin, 2019-04-15:
Auto-exposure is always enabled during the day (it does not depend on your settings). 

During the night, it starts using your settings so if you set auto-exposure to `off` it will apply the exposure value you entered in the text box and the exposure will remain the same during the whole night.

Kilis71, 2019-04-15:
Perfet, thanks.

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software and the request has been implemented in newer versions.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-05 by EricClaeys