AllskyTeam/allsky #84: Meteors search

Kilis71, 2019-04-19:
Hi Thomas, 
it would be very interesting to be able to add the meteors search function and create a photo of the trails, do you think it is possible to do it?
Thanks

ZebrasHaveBigEars, 2019-04-20:
Hi Thomas

I have nearly finished building my allsky which so far seems to work well. The meteor search function really does seem like an excellent idea. Cheers from Australia!

thomasjacquin, 2019-04-26:
Hi guys,

I've been thinking about it but haven't had the time yet to implement it.
I'm thinking about integrating this project with the allsky code: https://github.com/dwkim78/ASTRiDE

Kilis71, 2019-04-28:
Would be great.

Vincent-gits, 2019-05-06:
Sure it would be great !

ASTRIDE project seems to use FITS images, is it an issue ?

I just finished my Allsky (Pi zero W) using your software.
It works great !

thomasjacquin, 2019-05-06:
I believe you can convert JPG and PNG to fits before feeding it to ASTRIDE. Here's a quick example in Python:
http://docs.astropy.org/en/stable/generated/examples/io/split-jpeg-to-fits.html

ZebrasHaveBigEars, 2019-05-13:
It would be a great project ... I am not  sure I have the skills to usefully help though. I am happy to help test if this helps.

thomasjacquin, 2019-05-13:
I moved to my new house and I should have internet in a few days. Then I can put my camera back on line and start working on new features.

Chboss74, 2019-06-10:
Great idea, this was the reason to set up the allsky system in the beginning. Looking forward to hear about the progress and to test once ready.

ZebrasHaveBigEars, 2019-07-15:
I tried installing [ASTRiDE](https://github.com/dwkim78/ASTRiDE) but had no success. I think some of the installation routines may be a bit dated or something. Has anybody else had a go? With the new Raspberry 4 and the increase in processing power, I think the idea of meteor search becomes much more feasible.

Vincent-gits, 2019-08-06:
Hi there,

after a few days of search, yesterday I managed have ASTRiDE installed and running on a spare Pi3B+ and Stretch.

ASTRiDE manages to analyze my ASI120 B&W jpg/fits, but false detections are occuring with the landscape roof, trees,...  I need to play a bit more with all detection options and manage to have this landscape issue resolved.
And a nice integration to Allsky scripts/user interface will be required.

I will test with Pi4 and Buster too.

Vincent-gits, 2019-08-06:
[image]
[image]
[streaks.txt](https://github.com/thomasjacquin/allsky/files/3472192/streaks.txt)

ZebrasHaveBigEars, 2019-08-07:
Hi There Vincent

Might you be able to document or help others go down the ASTRiDE route especially if you can get it working on the Pi4 with Buster? As mentioned, I could not get everything to install after a day of trying. It would be magnificent if you could share! Thanks, ZHBE

Vincent-gits, 2019-08-08:
Here it is !

after Allsky installation into a fresh Buster Lite (2019-07-10-raspbian-buster-lite.img + Raspberry Pi4)

First install pip3
-----------------------------
sudo apt-get install python3-pip
Check using : pip3 --version

Install by hand required packages over Python3
------------------------------
pip3 install photutils
pip3 install scipy
pip3 install scikit-image
pip3 install matplotlib
sudo apt-get install libatlas-base-dev

Install ASTRiDE
------------------------------
git clone https://github.com/dwkim78/ASTRiDE
cd ASTRiDE
Modify setup.py : Remove the list of dependencies by hand. See attached file.
sudo /usr/bin/python3 setup.py install

Test ASTRiDE
--------------------------------
create file test.py :
*********
import matplotlib
matplotlib.use('Agg')

from astride import test
test()
***********

python3 test.py

Output files are into ~/ASTRiDE/astride/datasets/samples/long

If this last test is a success, congrats + you are ready for the next step...

Get into your Allsky jpg files
---------------------------------
Grab files attached Astride.py and autoDetect.sh

Usage : 
./manual_detect  /home/pi/allsky/images/   20190716   0100.jpg

Will test a single jpg image and create a dedicated folder for detected streaks..

./manual_detect  /home/pi/allsky/images/   20190716

Will test all jpg into the directory  and create a dedicated folder for detected streaks 

A lot to do....
------------------------------------
Add all the detection parameters available into Astride into theses scripts
Find the best detection parameters allowing best detection results
Find a way to remove the landscape from images...

Integrate this streaks detection feature into Allsky UI !

Add these scripts to Allsky/scripts/ repo ?
Add these packages dependencies to Allsky/install.sh ?

Known issue : If your jpg are not B&W then it might require modifications to Astride.py script...
See http://docs.astropy.org/en/stable/generated/examples/io/split-jpeg-to-fits.html as Thomas pointed earlier in this thread.

If some ppl play with this Astride tool, please share your results (good or bad !)

Cheers,
Vincent

[Astride.zip](https://github.com/thomasjacquin/allsky/files/3483727/Astride.zip)

ZebrasHaveBigEars, 2019-08-08:
Hi Vincent

This is truly heroic. Many thanks. I will have another go this weekend at installing ASTRiDE.

Double cheers!!!

ZHBE

ZebrasHaveBigEars, 2019-08-09:
Hi Vincent

Wahoo! Thanks to your terrific advice, I think it is installed. Scipy took two attempts but everything else went perfectly.  

Cheers!!!

ZHBE

Vincent-gits, 2019-08-12:
Hi ZHBE,

good to hear you managed to have it installed.
How are your detection tests ?

In order to ease testings, I will share a new script that handles all Astrid parameters through a specific JSON file.

ZebrasHaveBigEars, 2019-08-14:
Thanks Vincent. I have not actually tested it very much yet as this nasty thing called work has got in the way.

DarKOrange75, 2020-08-02:
Hi all,

has any progress been made on this subject?

Cheers.

WirthmU, 2020-08-07:
Same here - any news?

What about the JSON file Vincent-gits mentioned to handle ASTRiDE parameters?

Anyone figured out a way to handle non B&W jpg files?

DarKOrange75, 2020-08-13:
> Here it is !
> 
> after Allsky installation into a fresh Buster Lite (2019-07-10-raspbian-buster-lite.img + Raspberry Pi4)
> 
> ## First install pip3
> sudo apt-get install python3-pip
> Check using : pip3 --version
> 
> ## Install by hand required packages over Python3
> pip3 install photutils
> pip3 install scipy
> pip3 install scikit-image
> pip3 install matplotlib
> sudo apt-get install libatlas-base-dev
> 
> ## Install ASTRiDE
> git clone https://github.com/dwkim78/ASTRiDE
> cd ASTRiDE
> Modify setup.py : Remove the list of dependencies by hand. See attached file.
> sudo /usr/bin/python3 setup.py install
> 
> ## Test ASTRiDE
> create file test.py :
> 
> import matplotlib
> matplotlib.use('Agg')
> 
> from astride import test
> test()
> 
> python3 test.py
> 
> Output files are into ~/ASTRiDE/astride/datasets/samples/long
> 
> If this last test is a success, congrats + you are ready for the next step...
> 
> ## Get into your Allsky jpg files
> Grab files attached Astride.py and autoDetect.sh
> 
> Usage :
> ./autoDetect.sh /home/pi/allsky/images/ 20190716 0100.jpg
> 
> Will test a single jpg image and create a dedicated folder for detected streaks..
> 
> ./autoDetect.sh /home/pi/allsky/images/ 20190716
> 
> Will test all jpg into the directory and create a dedicated folder for detected streaks
> 
> ## A lot to do....
> Add all the detection parameters available into Astride into theses scripts
> Find the best detection parameters allowing best detection results
> Find a way to remove the landscape from images...
> 
> Integrate this streaks detection feature into Allsky UI !
> 
> Add these scripts to Allsky/scripts/ repo ?
> Add these packages dependencies to Allsky/install.sh ?
> 
> Known issue : If your jpg are not B&W then it might require modifications to Astride.py script...
> See http://docs.astropy.org/en/stable/generated/examples/io/split-jpeg-to-fits.html as Thomas pointed earlier in this thread.
> 
> If some ppl play with this Astride tool, please share your results (good or bad !)
> 
> Cheers,
> Vincent
> 
> [Astride.zip](https://github.com/thomasjacquin/allsky/files/3483727/Astride.zip)

Hi @Vincent-gits, thank you for this awesome job. I managed to install it (took me one day!) the test runs OK but when I try with one of my JPEG from the Raspberry HQ I get this error:

Traceback (most recent call last):
  File "Astride.py", line 99, in <module>
    main(sys.argv[1:])
  File "Astride.py", line 94, in main
    jpg2detect(inputfile,outputfile)
  File "Astride.py", line 60, in jpg2detect
    nb_data = nb_data.reshape(ysize, xsize)    # Reshape the image arrays to be 2-dimensional
ValueError: cannot reshape array of size 2250000 into shape (750,1000)

Any idea about how to fix it? Thank you.

jcauthen78, 2021-03-17:
Thanks for the hard work getting this info out to us - finally managed to get a Pi4 freshly imaged and decided to give this a shot. Short of the files within the zip.file throwing tons of errors (something about invisible carriage returns from some google searches _**($ bash: ./autoDetect.sh: /bin/bash^M: bad interpreter: No such file or directory)**_), fixed it with the '**dos2unix**' tool and they worked flawlessly (**_$ sudo apt-get install dos2unix_**).

I'm going to play around and see if i can figure out how to create a mask for the horizon lines/areas of no interest, and see if it'll run without detecting something(s) in every image at the horizons... maybe use some command-line tools from ImageMagick, not entirely sure yet. 

Has anyone found a suggested image dimension that is thought to work best? For now i ground the original images through ImageMagick to go from color --> BW & resize down to 800 height. the Astride script seemed to be ok with results (aside from the horizon lines).

EricClaeys, 2021-09-04:
Where does this issue stand?  Is it still being worked?  If not, would you please close it?  If it IS still being worked, is there an expected date to have it finished?
Thanks - Eric

linuxkidd, 2021-10-30:
This will be easy to add in a future release.  I'm working on a full replacement for all [post-processing in python](https://github.com/linuxkidd/allsky/blob/python-post/scripts/post-processor.py) that uses a [recipe](https://github.com/linuxkidd/allsky/blob/python-post/config_repo/postprocessing_night.json.repo) + [modules](https://github.com/linuxkidd/allsky/tree/python-post/scripts/modules) system.   This will look in two locations for modules, with the 'user' location overriding any distributed modules.

The meteor identification ( or anything else ) can be added to the pipeline.  Further, one recipe path can save a file, then the original can be reloaded for processing in an entirely different way, unchanged by the first path.

I plan to add this meteor identification as a system module, cause... it's super cool!  :D

EricClaeys, 2022-06-29:
Closing - this request was added to the Projects list - see the `Projects` link at the top of any Allsky GitHub page.
Project Title: **Meteors search - # 84**

closed 2022-06-29 by EricClaeys