AllskyTeam/allsky #85: Saving Fits? Minor typo

tdaiber, 2019-04-19:
Hello,

I too have all sky running perfect on a Pi Zero W 😃  It is just that creating the time-lapse takes hours. I think, I will relay this to my NAS, thereby freeing the Pi from the web server I am running right now.

Is it possible to add saving files as FITS format? Wouldn't the raw data be better for meteor trail frames?

While browsing the scripts to modify for my needs I found a minor typo: in endOfNight.sh the section creating the keogram is titled "Generate time-lapse..."

Great project, thanks a lot :grin:
Clear skies
Torsten

thomasjacquin, 2019-04-26:
Hi Torsten,

Thanks, I fixed the typo.

Time lapse can take a long time depending on the number of images. I've seen it take a couple of hours in the middle of winter with more than 6000 images.

Regarding the RAW data, if you set the image type to RAW8 or RAW16 and save the image as a PNG format, you should get the raw data without compression.
However, this will increase image size and probably slow down the timelapse even more.

EricClaeys, 2021-09-05:
This issue is from a much older release of the AllSky software and the question appears to have been answered.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks
Please note that saving .png files can take about 10 seconds so you need to make sure your exposure time + delay is at least 10 seconds or else you'll lose some frames.

closed 2021-09-05 by EricClaeys