AllskyTeam/allsky #86: segfaults

maphilli14, 2019-04-23:
Lately I've been getting many errors as follows:

Apr 15 20:49:45 allsky allsky.sh[1293]: convert-im6.q16: Premature end of JPEG file `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/352.
Apr 15 20:49:45 allsky allsky.sh[1293]: convert-im6.q16: Corrupt JPEG data: premature end of data segment `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/352.
Apr 15 21:03:51 allsky allsky.sh[1293]: scripts/saveImageNight.sh: line 43:  3714 Bus error               convert "$IMAGE_TO_USE" -resize 100x75 "images/$CURRENT/thumbnails/$FILENAME-$(date +'%Y%m%d%H%M%S').$EXTENSION"
Apr 15 21:08:52 allsky allsky.sh[1293]: convert-im6.q16: Premature end of JPEG file `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/352.
Apr 15 21:08:52 allsky allsky.sh[1293]: convert-im6.q16: Corrupt JPEG data: premature end of data segment `image.jpg' @ warning/jpeg.c/JPEGWarningHandler/352.

Apr 16 00:14:04 allsky allsky.sh[32713]: /home/pi/allsky/allsky.sh: line 25:   316 Segmentation fault      ./capture $ARGUMENTS

   Active: inactive (dead) (Result: exit-code) since Thu 2019-04-18 21:48:20 EDT; 1h 55min ago

   
   
   
Apr 22 00:19:18 allsky kernel: [13772.717867] usb 1-1.3: reset high-speed USB device number 4 using dwc_otg
Apr 22 00:19:27 allsky allsky.sh[25642]: scripts/saveImageNight.sh: line 43: 25774 Bus error               convert "$IMAGE_TO_USE" -resize 100x75 "images/$CURRENT/thumbnails/$FILENAME-$(date +'%Y%m%d%H%M%S').$EXTENSION"
Apr 22 01:09:44 allsky allsky.sh[25642]: /home/pi/allsky/allsky.sh: line 25: 25695 Segmentation fault      ./capture $ARGUMENTS

pi@allsky:~ $ sudo service allsky status
● allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-04-22 20:40:24 EDT; 1h 14min ago
 Main PID: 28326 (allsky.sh)
   CGroup: /system.slice/allsky.service
           ├─28326 /bin/bash /home/pi/allsky/allsky.sh
           └─28380 ./capture -angle -6 -autoexposure 1 -autogain 1 -bin 1 -brightness 50 -darkframe 0 -daytimeDelay 5

Apr 22 20:40:24 allsky systemd[1]: Started All Sky Camera.
pi@allsky:~ $ sudo /Scripts/freshenallsky.sh
pi@allsky:~ $ sudo service allsky status
● allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-04-22 21:55:08 EDT; 27s ago
 Main PID: 21320 (allsky.sh)
   CGroup: /system.slice/allsky.service
           ├─21320 /bin/bash /home/pi/allsky/allsky.sh
           └─21376 ./capture -angle -6 -autoexposure 1 -autogain 1 -bin 1 -brightness 50 -darkframe 0 -daytimeDelay 5

Apr 22 21:55:08 allsky systemd[1]: Started All Sky Camera.
pi@allsky:~ $
pi@allsky:~ $ df -h /
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        28G   18G  8.6G  68% /
pi@allsky:~ $ sudo service allsky status
● allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: exit-code) since Mon 2019-04-22 23:50:04 EDT; 26min ago
  Process: 21320 ExecStart=/home/pi/allsky/allsky.sh (code=exited, status=139)
 Main PID: 21320 (code=exited, status=139)

Apr 22 23:49:58 allsky systemd[1]: allsky.service: Failed with result 'exit-code'.
Apr 22 23:50:04 allsky systemd[1]: allsky.service: Service hold-off time over, scheduling restart.
Apr 22 23:50:04 allsky systemd[1]: Stopped All Sky Camera.
pi@allsky:~ $ sudo service allsky restart
pi@allsky:~ $

pi@allsky:~ $
pi@allsky:~ $ df -h /
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        28G   18G  8.6G  68% /
pi@allsky:~ $ sudo service allsky status
● allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: exit-code) since Mon 2019-04-22 23:50:04 EDT; 26min ago
  Process: 21320 ExecStart=/home/pi/allsky/allsky.sh (code=exited, status=139)
 Main PID: 21320 (code=exited, status=139)

Apr 22 23:49:58 allsky systemd[1]: allsky.service: Failed with result 'exit-code'.
Apr 22 23:50:04 allsky systemd[1]: allsky.service: Service hold-off time over, scheduling restart.
Apr 22 23:50:04 allsky systemd[1]: Stopped All Sky Camera.
pi@allsky:~ $ sudo service allsky restart
pi@allsky:~ $ sudo service allsky status
● allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: exit-code) since Tue 2019-04-23 00:19:10 EDT; 5min ago
  Process: 29198 ExecStart=/home/pi/allsky/allsky.sh (code=exited, status=139)
 Main PID: 29198 (code=exited, status=139)

Apr 23 00:19:04 allsky systemd[1]: allsky.service: Failed with result 'exit-code'.
Apr 23 00:19:10 allsky systemd[1]: allsky.service: Service hold-off time over, scheduling restart.
Apr 23 00:19:10 allsky systemd[1]: Stopped All Sky Camera.
pi@allsky:~ $

Last login: Mon Apr 22 20:33:33 2019 from 192.168.45.214
pi@allsky:~ $ sudo service allsky status
● allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: exit-code) since Tue 2019-04-23 03:02:29 EDT; 4h 9min ago
  Process: 30152 ExecStart=/home/pi/allsky/allsky.sh (code=exited, status=139)
 Main PID: 30152 (code=exited, status=139)

Apr 23 03:02:24 allsky systemd[1]: allsky.service: Main process exited, code=exited, status=139/n/a
Apr 23 03:02:24 allsky systemd[1]: allsky.service: Unit entered failed state.
Apr 23 03:02:24 allsky systemd[1]: allsky.service: Failed with result 'exit-code'.
Apr 23 03:02:29 allsky systemd[1]: allsky.service: Service hold-off time over, scheduling restart.
Apr 23 03:02:29 allsky systemd[1]: Stopped All Sky Camera.

00:05:29 allsky liblogging-stdlog:  [origin software="rsyslogd" swVersion="8.24.0" x-pid="317" x-info="http://www.rsyslog.com"] rsyslogd was HUPed
Apr 23 00:17:11 allsky kernel: [22594.397843] usb 1-1.3: reset high-speed USB device number 4 using dwc_otg
Apr 23 00:19:04 allsky allsky.sh[29198]: /home/pi/allsky/allsky.sh: line 25: 29240 Segmentation fault      ./capture $ARGUMENTS
Apr 23 00:24:42 allsky kernel: [23044.850971] usb 1-1.3: reset high-speed USB device number 4 using dwc_otg
Apr 23 00:24:56 allsky kernel: [23058.880994] usb 1-1.3: reset high-speed USB device number 4 using dwc_otg
Apr 23 00:31:04 allsky kernel: [23427.123158] Voltage normalised (0x00000000)
Apr 23 03:02:23 allsky allsky.sh[30152]: /home/pi/allsky/allsky.sh: line 25: 30209 Segmentation fault      ./capture $ARGUMENTS

linuxkidd, 2019-04-23:
Hello @maphilli14 
  It looks like the USB channel is getting reset to your camera which in turn, causes the allsky code to segfault.  I have two recommendations for that:

1: Try a different USB cable
 -- It's tough to find a quality one, and poor quality cables are the most common cause of repeated USB resets.

2: Make sure you have the latest version of the allsky code compiled on your Pi.
 -- There are some udev tweaks that will stop/start the allsky service when the camera disconnect/connects which may help prevent the 'segfault' and instead allow a graceful exit.

linuxkidd, 2019-04-24:
Hello @maphilli14 
  It looks like the USB channel is getting reset to your camera which in turn, causes the allsky code to segfault.  I have two recommendations for that:

1: Try a different USB cable
 -- It's tough to find a quality one, and poor quality cables are the most common cause of repeated USB resets.

2: Make sure you have the latest version of the allsky code compiled on your Pi.
 -- There are some udev tweaks that will stop/start the allsky service when the camera disconnect/connects which may help prevent the 'segfault' and instead allow a graceful exit.

Let us know if this helps.

maphilli14, 2019-04-24:
Thanks linuxkidd, I hope it's not the cable as it's tight in my tube.

linuxkidd, 2019-04-24:
@maphilli14 I'm afraid a hardware issue is most likely.  The only non-hardware contributor that could lead to USB resets is low input voltage on the RPi.  This could be worked around using a powered USB hub or just a better power supply for the RPi... 

But, USB cables are the most common cause of these resets by far.

Best of luck!

maphilli14, 2019-04-29:
I had been running a couple of scripts to create 'recent' animations https://f002.backblazeb2.com/file/allsky/LiveRefresh.html and after shutting them off it's been a lot more stable.  Also my Pi gets very hot and I have been shutting down during the day and powering off with PoE and turning back on with a fresh boot via PoE scheduling.  Since I'm at 35*N I power back up later.  I'm not sure what's causing it but seem not solely hardware

dbainbridge, 2019-04-29:
@maphilli14 The Pi will shutdown if it gets exceptionally hot.  There are a number or articles on how to cool it and heat issues with the Pi under heavy load: https://www.techrepublic.com/article/raspberry-pi-overheating-heres-how-to-stop-it-says-microsoft/

maphilli14, 2019-05-04:
I think my issues are related to what ever conflicts the scripts I was running were doing: https://github.com/maphilli14/allskysuppliments  You can close this, it's stable now.

closed 2019-05-04 by maphilli14